PlanSys2/ros2_planning_system #3: Implement clients in Python

fmrico, 2019-12-07:
We could use the planning system in Python, developing the clients also in Python.

iamshyamganatra, 2020-07-06:
I'm interested in working on this issue. Can I get more details about the python client that needs to be developed?

fmrico, 2020-07-08:
Hi @iamshyamganatra 

Great!! Adding the possibility of having client actions in Python would be an amazing contribution to PlanSys2.

As initial guidance, the idea is implementing the ActionExecutorClient class in Python:
- Each PDDL action in PlanSys2 is carried out by a ROS2 Node that implements a plansys2_msgs/action/ExecuteAction
- The class should hide most of the complexity of using actions.
- Use the terminal with "run (action_name arg1 arg2 ...)" for testing.

Start being familiar with the execution flow A++

iamshyamganatra, 2020-07-27:
Hi @fmrico , I am now familiar with this project, PDDL and got known to from how and where most of the things work. Now, can you tell me what things are needed to be done first?

P.S: I was busy with other things that, this went out of my mind.

fmrico, 2020-08-05:
Hi @iamshyamganatra, Yes, if you are familiar with Python, the support of Python, as this issue says, would be great!!

closed 2021-01-16 by fmrico