PlanSys2/ros2_planning_system #21: Support for Fluents and Metrics in the Package Expert

rossj13, 2020-05-08:
The support to dynamically added and remove fluents and goal metrics with the problem expert the same way predicates, instances and goals are would be very helpful

fmrico, 2020-05-11:
Hi @rossj13 

Thanks for your interest. I also think that fluents and metrics are an excellent feature to have. Popf supports it, but it should be managed from domain/problem experts, mainly.
PRs are also welcomed ;)

Best

teyssieuman, 2020-12-18:
@rossj13 : The fluents are now available : it is possible to assigne a function value from the problem expert client.
What do you mean by "goal metrics" ?

fmrico, 2020-12-20:
I agree with @rossj13 , it is necessary to address the goal metrics. 

@teyssieuman by goal metrics (@rossj13 correct me if I am wrong) I understand that the planner tries to optimize the minimization o maximization of a function when calculating a plan. From [here](http://users.cecs.anu.edu.au/~patrik/pddlman/writing.html):

> To specify action costs, it is necessary to add a "fluent" that keeps track of the cost. A fluent is like a state variable/predicate, but its value is a number instead of true or false. Again, note that PDDLs fluent syntax is very expressive, and most planners will only accept a limited use.
> To declare fluents, add the following section to the domain specification:
> 
> (:functions
>   (total-cost)
>  )
> Specifying that the cost of an action is C is done by stating that the action has the effect of increasing the total-cost fluent.
> 
>   :effect (and ... (increase (total-cost) C))
> The fluent must be initialised. In the :init section of the problem definition, add
> 
>   (= (total-cost) 0)
> Finally, to specify that minimising the sum of action costs is the objective, a :metric section is added to the problem definition:
> 
> (define (problem ..)
>   .
>   .
>   .
>   (:metric minimize (total-cost))
>  )

fmrico, 2021-01-16:
Hi @rossj13 

Since #62 , this should be fixed. Can you test it so we can close this issue?

Thanks!!

rossj13, 2021-01-25:
@fmrico sorry for the delay. Ill try the new changes

rossj13, 2021-02-16:
Looks like the fluents work. Still needs support for metrics to maximize the fluents usefulness.

sethhoward, 2021-03-31:
Metrics would be a huge win for some of our use cases as well as actor selection.

But I think the bigger issue is none of the plugins (POPF, Fast Downward, Mips, LPG) support metrics from my testing.

fmrico, 2021-04-04:
Hi @sethhoward 

My students analyzed in one exercise a lot of planner, and they found some interesting planners that supported metrics. Let me check the results because they also created plugins for most of them.

Best

kogut, 2021-04-28:
> Metrics would be a huge win for some of our use cases as well as actor selection.
> 
> But I think the bigger issue is none of the plugins (POPF, Fast Downward, Mips, LPG) support metrics from my testing.

I can corroborate popf, TFD, and MIPS-XXL  (the versions indicated by PlanSys2 plugins).

I've had a bit of luck with metrics in LPG-TD.  (in fact I can't seem to successfully plan without one which makes it hard to run baseline tests).   I created a simple "energy-used" metric with a function keeps track of energy subtracted from the battery.   When running with the metric, the LPG-TD planner results indicate that the metric is being used both in the "C" field of each action as well as the total "Execution cost" reported to stdout on the console.    What I haven't been able to demonstrate is that LPG-TD is using the metric in any meaningful way.   I see little difference when switching between "minimize" and "maximize."  And it doesn't help that LPG-TD has a surprising amount of variation in plan results in repeated trials (unlike popf which is super deterministic).  

I'll report new findings here since I share this interest.

kogut, 2021-05-12:
> But I think the bigger issue is none of the plugins (POPF, Fast Downward, Mips, LPG) support metrics from my testing.

So far I've found that KCL's Optic planner (a derivative of popf) provides really good support for metrics that use numeric fluents. 

There, unfortunately, is no really good git repo with a version that's conveniently build-able using a modern build system (like @fmrico did for popf).  But it only took me a few hours from the 2012 optic tarball from KCL.  I may work on a PlanSys2 plugin for optic as well as a ROS 2-build-friendly version of optic, but no promises yet.

fmrico, 2021-05-18:
@kogut, Maybe using the binary is not so bad...

roveri-marco, 2021-06-01:
I've successfully compiled latest version of optic on Ubuntu 20.04 LTS. 
You can find it here: https://github.com/roveri-marco/optic

fmrico, 2021-06-02:
@roveri-marco @kogut 

Excellent!! The next step is having a planner plugin similar to the ones for [popf](https://github.com/IntelligentRoboticsLabs/ros2_planning_system/tree/master/plansys2_popf_plan_solver) or [lpgtd](https://github.com/Nuriadj/plansys2_lpgtd_plugin). These plugins only call to the binary and parse the output. You could add in this plugin a reference to https://github.com/roveri-marco/optic in the building instructions.

We should be thinking about supporting metrics at runtime in PlanSys2. @jjzapf, I think that this is not covered in #118, isn't it?

Thanks!

kogut, 2021-06-02:
I will create the planner plugin, and work with @jjzapf on proper metric support throughout PlanSys2 (not covered by #118).

kogut, 2021-06-14:
@fmrico - quick question.  I have completed a cleanup and port of optic to build natively within a ROS 2 workspace (as in popf).  I am working on approval to make that repository public-facing.    I am now working on the plugin, which should be reasonably straightforward.  Should I 1) make the plugin part of this repo, maybe adding 'plansys2_plugins' and placing it parallel to popf within the new directory, 2) make it a standalone repository like some of the other plugins, and add it to the "approval list".

kogut, 2021-06-15:
I have completed both an optic ROS 2 package (similar to popf's ROS package) and a PlanSys2 plugin for that package.  I'll post the github URLs here once I get approval for making the code public.

gregtkogut, 2021-09-02:
The OPTIC plugin is now operational in the same repo as above.     However it's only useful for simple plans at the moment because of a strict service call timeout within the PlanSys2.  I'm working on that in another issue, as a pull request to this repo.

fmrico, 2021-09-16:
@gregtkogut Maybe we could parametrize this parameter, and adding it to the config file.

gregtkogut, 2021-09-16:
Yes, that's my intention.  I have a branch that does that, that I'm testing.

fmrico, 2022-07-09:
Closing... reopen if you want to continue with this

closed 2022-07-09 by fmrico

oyssolbo, 2023-03-15:
@gregtkogut Is there any progress on the support for metrics or OPTIC?

gregtkogut, 2023-03-15:
The above GitHub link has a version of OPTIC that with modern compilation and PlanSys2 integration.  I know it's building (part of a nightly build), but I haven't actually used the plugin in quite some time.

OPTIC metrics:  As a standalone planner, I found OPTIC to have some support for metrics, but it seems fragile, narrow, and inconsistent in important ways.  I believe some "refactoring" would be necessary for most metric use.  The original developers have done this, I believe, but have made that proprietary code.

PlanSys2 metrics:  I haven't checked recently, but I don't believe PlanSys2 yet has native support for metrics.  It would require a lot of work to do right, outside the scope of my personal time.

fmrico, 2023-03-18:
All depends on popf, as far as I remember. popf accepts metrics, but it doesn't really use it.