PlanSys2/ros2_planning_system #35: Support for additional constructs in the domain expert

jjzapf, 2020-11-19:
The pddl parser supports more PDDL constructs than are in the domain expert. To support these additional constructs, I believe some changes are needed to the way in which predicate trees are created. Currently, PDDLPrint() functions in the pddl parser are called to return strings to the domain expert. These strings are then converted to predicate trees by the domain expert. This intermediary string conversion seems unnecessary. In addition, it can be difficult to determine the appropriate construct from a string snippet alone. For example "(vehicle_charge ?r)" could be a predicate or a function.

I propose that the code encapsulated by the Types.hpp/cpp files in the domain expert be moved to the pddl parser in new files called Tree.h/cpp. We could then add PDDLTree() functions to the various pddl construct classes in the pddl parser. These functions would directly compute the predicate trees from the pddl parser objects rather than use the intermediary string conversions.

fmrico, 2020-11-27:
Hi @jjzapf 

I am not very happy with the result either for the reason you indicate. We could have more additional builds without this intermediate step. As I told you in the mail, this has been done to be able to merge several PDDL domains, but possibly there is some other alternative. Maybe merge the domains and then use the PDDL Parser.

Do you think that you could do a PR with your proposed solution?

Thanks!!

jjzapf, 2020-11-30:
Thank you for your reply! Yes, I am currently working on the changes and my organization will need to approve the release. Hopefully I will be able to open a PR sometime in December.

jjzapf, 2020-12-12:
Hello @fmrico

I just wanted to give you a heads up that our internal review process for these changes is nearly complete. Once this is done, I will submit a diff to my organization as part of a request to release this work. I was planning on making a pull request to the master branch. However, I could also make a PR the foxy-devel branch with a small amount of work. What is your preferred integration strategy.

Thanks!

fmrico, 2020-12-14:
Hi @jjzapf 

I prefer a PR to the master branch. Don't forget to add some tests ;)

Thanks!!

fmrico, 2021-01-16:
Hi @jjzapf , I am closing this, as fixed in #62 . Please, reopen or open a new issue if you find more problems

closed 2021-01-16 by fmrico