PlanSys2/ros2_planning_system #37: I would like to base my contibutions on the master branch

teyssieuman, 2020-12-04:
The master branch seems not to be in a working state.

I'm using the current 'master' branch and the 'behavior_tree_executor' branch of the 'example' depot.

In a first terminal : 
```
ros2 launch plansys2_simple_example plansys2_simple_example_launch.py
[INFO] [launch]: All log files can be found below /home/sdri1/.ros/log/2020-12-04-11-52-19-560180-sdri1-Alienware-Area-51m-R2-915048
[INFO] [launch]: Default logging verbosity is set to INFO
/home/sdri1/ros2_ws/install/share/plansys2_bringup/launch/plansys2_bringup_launch_monolithic.py:47: UserWarning: The parameter 'node_executable' is deprecated, use 'executable' instead
  plansys2_node_cmd = Node(
/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py:184: UserWarning: The parameter 'node_namespace' is deprecated, use 'namespace' instead
  warnings.warn("The parameter 'node_namespace' is deprecated, use 'namespace' instead")
[INFO] [plansys2_node-1]: process started with pid [915050]
[INFO] [move_action_node-2]: process started with pid [915052]
[INFO] [charge_action_node-3]: process started with pid [915054]
[INFO] [ask_charge_action_node-4]: process started with pid [915056]
[move_action_node-2] RCUTILS_CONSOLE_STDOUT_LINE_BUFFERED is now ignored.  Please set RCUTILS_LOGGING_USE_STDOUT and RCUTILS_LOGGING_BUFFERED_STREAM to control the stream and the buffering of log messages.
[plansys2_node-1] RCUTILS_CONSOLE_STDOUT_LINE_BUFFERED is now ignored.  Please set RCUTILS_LOGGING_USE_STDOUT and RCUTILS_LOGGING_BUFFERED_STREAM to control the stream and the buffering of log messages.
[charge_action_node-3] RCUTILS_CONSOLE_STDOUT_LINE_BUFFERED is now ignored.  Please set RCUTILS_LOGGING_USE_STDOUT and RCUTILS_LOGGING_BUFFERED_STREAM to control the stream and the buffering of log messages.
[ask_charge_action_node-4] RCUTILS_CONSOLE_STDOUT_LINE_BUFFERED is now ignored.  Please set RCUTILS_LOGGING_USE_STDOUT and RCUTILS_LOGGING_BUFFERED_STREAM to control the stream and the buffering of log messages.
[plansys2_node-1] [INFO] [1607079139.655475688] [domain_expert_lc_mngr]: Creating client for service [domain_expert/get_state]
[plansys2_node-1] [INFO] [1607079139.655559916] [domain_expert_lc_mngr]: Creating client for service [domain_expert/change_state]
[plansys2_node-1] [INFO] [1607079139.656017956] [executor_lc_mngr]: Creating client for service [executor/get_state]
[plansys2_node-1] [INFO] [1607079139.656031558] [executor_lc_mngr]: Creating client for service [executor/change_state]
[plansys2_node-1] [INFO] [1607079139.656488483] [planner_lc_mngr]: Creating client for service [planner/get_state]
[plansys2_node-1] [INFO] [1607079139.656500412] [planner_lc_mngr]: Creating client for service [planner/change_state]
[plansys2_node-1] [INFO] [1607079139.656957213] [problem_expert_lc_mngr]: Creating client for service [problem_expert/get_state]
[plansys2_node-1] [INFO] [1607079139.656968771] [problem_expert_lc_mngr]: Creating client for service [problem_expert/change_state]
[plansys2_node-1] [INFO] [1607079139.658305359] [domain_expert]: [domain_expert] Configuring...
[plansys2_node-1] [INFO] [1607079139.659078129] [domain_expert]: [domain_expert] Configured
[plansys2_node-1] [INFO] [1607079139.659400426] [domain_expert_lc_mngr]: Transition 1 successfully triggered.
[plansys2_node-1] [INFO] [1607079139.659790848] [domain_expert_lc_mngr]: Node domain_expert_lc_mngr has current state inactive.
[plansys2_node-1] [INFO] [1607079139.659982655] [problem_expert]: [problem_expert] Configuring...
[plansys2_node-1] [INFO] [1607079139.660340062] [problem_expert]: [problem_expert] Configured
[plansys2_node-1] [INFO] [1607079139.660559987] [problem_expert_lc_mngr]: Transition 1 successfully triggered.
[plansys2_node-1] [INFO] [1607079139.660893872] [problem_expert_lc_mngr]: Node problem_expert_lc_mngr has current state inactive.
[plansys2_node-1] [INFO] [1607079139.661032002] [planner]: [planner] Configuring...
[plansys2_node-1] [INFO] [1607079139.661409825] [planner]: Created solver : POPF of type plansys2::POPFPlanSolver
[plansys2_node-1] [INFO] [1607079139.661462260] [planner]: [planner] Configured
[plansys2_node-1] [INFO] [1607079139.661951895] [planner_lc_mngr]: Transition 1 successfully triggered.
[plansys2_node-1] [INFO] [1607079139.662166191] [planner_lc_mngr]: Node planner_lc_mngr has current state inactive.
[plansys2_node-1] [INFO] [1607079139.662249136] [executor]: [executor] Configuring...
[plansys2_node-1] [INFO] [1607079139.670070415] [executor]: [executor] Configured
[plansys2_node-1] [INFO] [1607079139.670578698] [executor_lc_mngr]: Transition 1 successfully triggered.
[plansys2_node-1] [INFO] [1607079139.670850270] [executor_lc_mngr]: Node executor_lc_mngr has current state inactive.
[plansys2_node-1] [INFO] [1607079139.670978290] [domain_expert]: [domain_expert] Activating...
[plansys2_node-1] [INFO] [1607079139.670987628] [domain_expert]: [domain_expert] Activated
[plansys2_node-1] [INFO] [1607079139.671080919] [domain_expert_lc_mngr]: Transition 3 successfully triggered.
[plansys2_node-1] [INFO] [1607079139.671154595] [problem_expert]: [problem_expert] Activating...
[plansys2_node-1] [INFO] [1607079139.671163278] [problem_expert]: [problem_expert] Activated
[plansys2_node-1] [INFO] [1607079139.671249269] [problem_expert_lc_mngr]: Transition 3 successfully triggered.
[plansys2_node-1] [INFO] [1607079139.671322998] [planner]: [planner] Activating...
[plansys2_node-1] [INFO] [1607079139.671330278] [planner]: [planner] Activated
[plansys2_node-1] [INFO] [1607079139.671362014] [planner_lc_mngr]: Transition 3 successfully triggered.
[plansys2_node-1] [INFO] [1607079139.671436260] [executor]: [executor] Activating...
[plansys2_node-1] [INFO] [1607079139.671443531] [executor]: [executor] Activated
[plansys2_node-1] [INFO] [1607079139.671474692] [executor_lc_mngr]: Transition 3 successfully triggered.
[plansys2_node-1] [INFO] [1607079139.671622379] [domain_expert_lc_mngr]: Node domain_expert_lc_mngr has current state active.
[plansys2_node-1] [INFO] [1607079139.671716168] [problem_expert_lc_mngr]: Node problem_expert_lc_mngr has current state active.
[plansys2_node-1] [INFO] [1607079139.671809863] [planner_lc_mngr]: Node planner_lc_mngr has current state active.
[plansys2_node-1] [INFO] [1607079139.671956221] [executor_lc_mngr]: Node executor_lc_mngr has current state active.
```

In a second one:
```
ros2 run plansys2_terminal plansys2_terminal        # enters in PlanSys2 Terminal
ROS2 Planning System console. Type "quit" to finish
> set instance leia robot
> set instance entrance room
[... 47 lines cut ...]
(charging_point_at chargingroom)
(battery_low leia)
(robot_at leia entrance)
> 
> set goal (and(robot_at leia bathroom))                # Sets the goal
> get plan                                              # Creates plan and shows it
plan: 
0	(askcharge leia entrance chargingroom)	5
0.001	(charge leia chargingroom)	5
5.002	(move leia chargingroom kitchen)	5
10.003	(move leia kitchen dinning)	5
15.004	(move leia dinning bedroom)	5
20.005	(move leia bedroom bathroom)	5
```

Then, in the second terminal :
```
> run

```
Cause in the first one:
```[plansys2_node-1] terminate called after throwing an instance of 'BT::RuntimeError'
[plansys2_node-1]   what():  Possible typo? In the XML, you tried to remap port "threshold" in node [Parallel / Parallel], but the manifest of this node does not contain a port with this name.
[ERROR] [plansys2_node-1]: process has died [pid 915050, exit code -6, cmd '/home/sdri1/ros2_ws/install/lib/plansys2_bringup/plansys2_node --ros-args -r __ns:=/ --params-file /tmp/launch_params_9zwe0x00 --params-file /home/sdri1/ros2_ws/install/share/plansys2_bringup/params/plansys2_params.yaml'].
```

teyssieuman, 2020-12-04:
As I said, I would like to base my future developments on the master branch, so, how can I contribute to make it works ?

fmrico, 2020-12-08:
Hi @teyssieuman 

My fault. I didn't finally merge https://github.com/IntelligentRoboticsLabs/ros2_planning_system_examples/pull/5

Try again, please. It is working for me.

Best

teyssieuman, 2020-12-08:
I just checkout and compile a brand new couple of  `ros2_planning_system` / `ros2_planning_system_examples` but I still have the same issue :
```
[plansys2_node-1] plan: 
[plansys2_node-1] 0	(askcharge leia entrance chargingroom)	5
[plansys2_node-1] 0.001	(charge leia chargingroom)	5
[plansys2_node-1] 5.002	(move leia chargingroom kitchen)	5
[plansys2_node-1] 10.003	(move leia kitchen dinning)	5
[plansys2_node-1] 15.004	(move leia dinning bedroom)	5
[plansys2_node-1] 20.005	(move leia bedroom bathroom)	5
[plansys2_node-1] terminate called after throwing an instance of 'BT::RuntimeError'
[plansys2_node-1]   what():  Possible typo? In the XML, you tried to remap port "threshold" in node [Parallel / Parallel], but the manifest of this node does not contain a port with this name.
[ERROR] [plansys2_node-1]: process has died [pid 26352, exit code -6, cmd '/home/sdri1/test_ws/install/plansys2_bringup/lib/plansys2_bringup/plansys2_node --ros-args -r __ns:=/ --params-file /tmp/launch_params_tpf3f8le --params-file /home/sdri1/test_ws/install/plansys2_bringup/share/plansys2_bringup/params/plansys2_params.yaml'].
```

fmrico, 2020-12-08:
Hi @teyssieuman 

Yes, now I am able to reproduce the error. I had the behavior_tree package from sources in my directory (not updated) and that masked the error for me.

Let me check if I can easily fix it

fmrico, 2020-12-08:
Fixed in #39 

Thanks for warning this!!

PR #39 (2020-12-08): Fix threshold input in parallel nodes
This PR fixes a change in the Behavior Tree library in the parallel node. Now It is necessary to specify `success_threshold` and `failure_threshold`.

Fixes #37
closed 2020-12-08 by fmrico