PlanSys2/ros2_planning_system #38: Error adding predicates after actions

dr-steveb, 2020-12-04:
The system seems to produce an error `Action client execution error applying at_end effects` if you apply a predicate that is already true. This can easily be seen with the terminal tool and the simple example (simple_example.pddl). From the tool, type `set predicate (robot_at leia entrance)` twice. The second time produces the error `Predicate not valid`. If this happens during the actual execution of the plan, the execution will abort. Is this intentional? Is there any way to allow true predicates to be set anyway?

fmrico, 2020-12-08:
Hi @dr-steveb 

That behavior is not correct and I thought I had fixed it some time ago...

I can fix it, but not immediately. Meanwhile, if you want to fix it, a PR is welcome ;)

Best

dr-steveb, 2020-12-08:
Hi,
Which branch should I be looking at for this? Note that I am using Foxy. Happy to fix if I can.

Best,
Stephen

fmrico, 2020-12-14:
Hi @dr-steveb 

The development should be always done in the master branch. Foxy or other distribution-specific branches are used only to deliver packages.

Thanks!!

dr-steveb, 2020-12-16:
Created a bug fix branch off of master for this and fixed issue. However, I can't push the new branch to the repo. How do I get permissions?

fmrico, 2020-12-17:
I don't know if you used to follow the usual open-source contribution procedure. If not, these are the steps:
1. Fork PlanSys2 repo (this repo) in your account
2. Clone your repo (or add a remote) in your computer.
3. Create a branch with your contribution.
4. Push this branch to your forked repo.
5. Do a Pull Request (PR) to PlanSys2 repo.
6. Then we could start talking, reviewing, ... 
7. When it is finished, I will merge your PR in the PlanSys2 repository.

fmrico, 2021-01-16:
@dr-steveb I am closing this, as fixed in #51

closed 2021-01-16 by fmrico