PlanSys2/ros2_planning_system #46: Is the actions durrations time quantum 1s ?

teyssieuman, 2020-12-15:
https://github.com/IntelligentRoboticsLabs/ros2_planning_system/blob/2365ac8e5072fda927cfe987a5aa252884379158/plansys2_executor/src/plansys2_executor/BTBuilder.cpp#L328-L331

It seems that the actions duration is converted to "integer" on purpose (it do not seems to be a bug). 

POPF support millisecond resolutions in the plan.

So, is this a reason to keep a non floating-point value for the actions duration ? is this possible to keep an integer value but, for example, an integer value of milliseconds ?

fmrico, 2020-12-16:
When creating the planning graph, I assume that time is integer to identify the actions that are planned at the same time, or approximately. You can see it in this image:

[image]

For example:

1.01 action1
1.02 action2
2.01 action3

action1 and action2 are supposed to happen at the same time and action3 in the next time. 

Maybe it can be improved, but this is the algorithm. BTW, this is written in a paper waiting for review.

teyssieuman, 2020-12-17:
Ok, that make sense.

But why does the tasks have to be grouped in such a way in order to be executed ? Is this a simplification for the plan execution algorithm ?

Is this ok for the above case where a task last less than 1 s ?

[image]

fmrico, 2020-12-20:
Hi @teyssieuman 

I sent you a couple of days ago an email with the paper with the description of this algorithm (BTW, just accepted in AAMAS2021). We can discuss here any improvement to the algorithm. 

In general, I didn't consider any action with a duration of less than 1 second, that's why I used integers. This simplified me the implementation.

teyssieuman, 2020-12-20:
Hello @fmrico I just started to read the paper you sent me, but my holidays do not leave me so much time (strange but true)...

In the case of my application, I have some actions that lasts less than 1 seconds (like 100 ms).

I can try to submit some modification of the current implementation. Do you prefer converting it to floating point or do you thing it is better to still use fix point ? and in this last case, what time quantum to use ? should is be this configurable ?

fmrico, 2021-01-16:
Since #72 , there are not execution levels and the time is not truncated to seconds, so I am closing this

closed 2021-01-16 by fmrico

teyssieuman, 2021-01-16:
Perfect ...