PlanSys2/ros2_planning_system #47: Correct the indentation in order to make the CI work.

teyssieuman, 2020-12-15:
https://github.com/IntelligentRoboticsLabs/ros2_planning_system/blob/2365ac8e5072fda927cfe987a5aa252884379158/plansys2_executor/src/plansys2_executor/BTBuilder.cpp#L78

teyssieuman, 2020-12-15:
```
  [3.464s] 7: --- src/plansys2_executor/BTBuilder.cpp
  [3.464s] 7: +++ src/plansys2_executor/BTBuilder.cpp.uncrustify
  [3.464s] 7: @@ -78 +78 @@
  [3.464s] 7: -        "\" failure_threshold=\"1\">\n";
  [3.464s] 7: +      "\" failure_threshold=\"1\">\n";
  [3.464s] 7: @@ -215 +215 @@
  [3.464s] 7: -        "\"  failure_threshold=\"1\">\n";
  [3.464s] 7: +      "\"  failure_threshold=\"1\">\n";
```

According to uncrustify, the insentation shall be corrected.

fmrico, 2020-12-16:
Hi @teyssieuman 

Yes, I have noticed this today. My local tests are different from the tests in CI. I prefer to make CI happy :)

fmrico, 2020-12-16:
Actually, it's been included in #45

teyssieuman, 2020-12-17:
Perfect 👼

closed 2020-12-17 by teyssieuman