PlanSys2/ros2_planning_system #57: PlanSys2 Web page

fmrico, 2020-12-23:
Dear users and contributors,

I have just created the initial version of the [PlanSys2 web page](http://intelligentroboticslab.gsyc.urjc.es/ros2_planning_system.github.io/).

Feel free to contribute to it with docs and tutorials, through its [Github repo](https://github.com/IntelligentRoboticsLabs/ros2_planning_system.github.io).

The steps to build the web locally is:
1. Clone the repo
```
git clone https://github.com/IntelligentRoboticsLabs/ros2_planning_system.github.io.git
```
2. Make the changes using [RST](https://thomas-cokelaer.info/tutorials/sphinx/rest_syntax.html)
3. Build the web by running in the root directory of the project:
```
make html
```
4. Verify the result
```
firefox _build/html/index.html
```
5. When you are happy with the result, make a PR as usual.

Happy editing!!!

sethhoward, 2020-12-24:
I would like to contribute documentation and an example of multi-robot planning after the holidays. We're looking at deploying and testing in simulation in the beginning of the year before deploying to physical robots in a controlled environment.

Because there isn't any documentation around this I wanted to reach out and make sure the way we've implemented multi-bot planning is as intended:

We load a name spaced yaml much like the one found in the `bt_example /config/params.yaml` for each robot.

```
/robot1:
    move:
      ros__parameters:
        plugins:
       ...
```

We create action nodes using `specialized_arguments` and found this does the trick when having many robots in a scene and issuing orders to just one or creating a problem domain that has several carry out the plan in parallel.

```    
       move_cmd = Node(
        package='plansys2_bt_actions',
        executable='bt_action_node',
        name='move',
        namespace=namespace,
        output='screen',
        parameters=[
          [planner_dir, '/params/plansys_params_', namespace, '.yaml'],
          {
            'specialized_arguments': [[namespace]],
            'action_name': 'move',
            'bt_xml_file': example_dir + '/behavior_trees_xml/move.xml'
          }
        ])
```

Before I continue work on this example and documentation, I want to be sure this is how multi-robot missions are intended to be initialized.

Thanks for your time and all your work!

fmrico, 2020-12-25:
Hi @sethhoward 

Exactly!! The purpose of `specialized_arguments` is to have a mechanism to select among more than one action performer for the same action. Two applications are:
- Multirobot: That's why I used to include the robot as an argument in actions (the subject). Each robot can have nodes to perform the same actions and be responsible only to its id by this mechanism.
- Specialized performers: Yo can have different implementations for the same action and select among using this mechanism.

While ROSPlan uses ROS actions to deliver the action from the planning system to the action implementations, PlanSys2 uses a different approach based on bidding. A protocol uses the `/actions_hub` topic to request an action performer at runtime. The action performer that "feels" fit in the request (taking into account the action name and `specialized_arguments`) responds to this request. Once one of them is confirmed, it starts to execute.

I will be documenting all of this these weeks, but I am impressed that you found out how to do multirobot only by watching the code.

closed 2020-12-30 by fmrico