PlanSys2/ros2_planning_system #61: PDDL 'assign' keyword not supported

mark-sentaca, 2020-12-29:
Relatively new to PDDL here, but as far as we've researched, the 'assign' token is a well-supported standard, and verifies under popf.

In our PDDL domain, we're using something like:

```
     (at end(assign (cargo ?r ?p) 0))
```
     
which generates an error of the following form:

```
[plansys2_node-1] [ERROR] [1609280626.223540757] []: Original error: 'assign' does not name a known token
```

I'm assuming the error is in the bison parser support?  We're an organization that is willing to invest some time to submit a PR to an open source project like this, but at this point I'm not sure where to start or whether there's another, better approach.

fmrico, 2021-01-16:
Hi @mark-sentaca 

Could we close this issue? Did you check in the current main?

Thanks!!

mark-sentaca, 2021-01-20:
Hello Francisco,

It didn't get resolved but it can be closed since we're not working along
this path currently.  I was still not seeing the keyword supported, but
it's not on any critical path for us at the moment.

Mark

closed 2021-01-22 by fmrico