PlanSys2/ros2_planning_system #69: No action performer when trying to move two bots

sethhoward, 2021-01-11:
Everything is working fine until commit fcf5aba5787cdc7f94cc2ccb2c6c1301d7dcd8e9

As mentioned previously, I'm working on a demo for multiple robots based off of the bt_example. Everything was working as expected until the aforementioned commit. Planner returns a plan but when run, one of the bots begins execution while the other logs `No action performer for (move bot assembly_zone body_car_zone). retrying` until eventual failure.

fmrico, 2021-01-15:
Hi @sethhoward 

You should have two instances of the `move` node, like this:

```
    move_1_cmd = Node(
        package='plansys2_bt_actions',
        executable='bt_action_node',
        name='move_1',
        namespace=namespace,
        output='screen',
        parameters=[
          example_dir + '/config/params.yaml',
          {
            'action_name': 'move',
            'bt_xml_file': example_dir + '/behavior_trees_xml/move.xml'
          }
        ])

    move_2_cmd = Node(
        package='plansys2_bt_actions',
        executable='bt_action_node',
        name='move_2',
        namespace=namespace,
        output='screen',
        parameters=[
          example_dir + '/config/params.yaml',
          {
            'action_name': 'move',
            'bt_xml_file': example_dir + '/behavior_trees_xml/move.xml'
          }
        ])
```
It is supposed that one is busy executing one action. So you need another. In a real application, each robot has its own move node.

sethhoward, 2021-01-15:
Ok, I'll give that a try. Previously this worked:

    move_cmd = Node(
        package='plansys2_bt_actions',
        executable='bt_action_node',
        name='move',
        namespace=namespace,
        output='screen',
        parameters=[
          [planner_dir, '/params/plansys_params_', namespace, '.yaml'],
          {
            'specialized_arguments': [[namespace]],
            'action_name': 'move',
            'bt_xml_file': example_dir + '/behavior_trees_xml/move.xml'
          }
        ])

Where this node was generated for each robot being brought into the world. I'm assuming this was just a lucky coincidence and not the proper creation and the `name` as well as `specialized_arguments` should be unique for each bot?

fmrico, 2021-01-16:
Hi @sethhoward 

First of all, update to HEAD as soon as I merge #73. This PR contains a utility to check the action performers for each action. This PR also fixes a possible bug I introduced some commits ago. I changed the `/actions_hub` topic to `actions_hub`, but it could be a failure if you use a different namespace for each robot, so I have just reverted.

If you don't find section performers, check that they exist and they are in an inactive state. Maybe someone didn't configure correctly.

I would like also to explain the protocol to find the action performers, so you can understand the problem:
* I use the topic `/actions_hub` to deliver the actions to each action performer, which is implemented in `plansys::ActionExecutorClient`. The other part of this communication is in the Executor, instantiating a `plansys::ActionExecutor` for each action to execute.
* You can echo `/actions_hub` to see the `plansys2_msgs/msg/ActionExecution` messages. This message implements the protocol.
* Each time an action has to be executed this happens:
  1. The Executor sends a `plansys2_msgs::msg::ActionExecution::REQUEST` with the action and the parameters, asking for any action performer to execute it.
  2. As soon as each action performer receives the request, each action performer checks if it implements that action, is inactive and the arguments fit (here the `specialized_arguments`, if any,  are checked). If affirmative, it sends a `plansys2_msgs::msg::ActionExecution::REQUEST`, declaring that it is willing to execute. Here, this action performed is committed to this request. This avoids sending more than one response for different requests.
  3. The Executor receives the responses. The first response is confirmed with a `plansys2_msgs::msg::ActionExecution::CONFIRM` message. Successive responses are answered with `plansys2_msgs::msg::ActionExecution::REJECT`.
  4. When an action performer receives a `plansys2_msgs::msg::ActionExecution::CONFIRM`, it activates to execute the action. In case of a rejection, it releases the commit, and it is ready to participate in another request.

I will write this description in the documentation on the web page soon, but I wanted to give you more information now.

I hope this helps you :wink:

sethhoward, 2021-01-28:
Thank you, this latest commit resolved the issue we were seeing, solved a couple edge cases in the PDDL but introduced something new.

In the foxy release we run multiple bots with PDDL declared a bit like:
```

(:durative-action move
    :parameters (?r - robot ?z1 ?z2 - zone)
    :duration ( = ?duration 5)
    :condition (and
        (at start(robot_at ?r ?z1)))
    :effect (and
        (at start(not(robot_at ?r ?z1)))
        (at end(robot_at ?r ?z2))
    )
)

(:durative-action pickup
    :parameters (?r - robot ?p - piece ?z - zone)
    :duration ( = ?duration 5)
    :condition (and
        (at start(robot_at ?r ?z))
        (over all (robot_at ?r ?z))
        
        (at start(piece_at ?p ?z))
    )
    :effect (and
        (at end(carry ?r ?p))
        (at end (not (piece_at ?p ?z)))
        (at start (not(piece_not_used ?p)))

        ; also, Plansys2 skill layer will fail without this. In particular, it will not engage the pick and place action (open, approach, close).
        (at start(robot_at ?r ?z))
    )
)
```
We're running the POPF executable through a VSCode plugin so we can view the plan and swim lanes before executing in Plansys2. We've noticed odd side effects when engaging the skill layer with our plan (below) in Plansys2.

We've separated the pick and place (previously `transport`) behavior trees into their own nodes (pickup, dropoff) and continue to use the `move` and `assembly` node/pddl unchanged. You can see in our PDDL there's a comment that not declaring the `robot_at` effect `at start` results in no action/execution by the BT pickup, same applies to dropoff (eg. no print out that the robot is opening the gripper, approaching the object, etc. As well as no delay in execution). 

The latest commits to master have changed this so it's no longer necessary to declare the `robot_at` effect in the above PDDL. Instead what we're seeing is the `assembly` action fires after two bots delivery the first two items but before the third is picked up and dropped off. The plan looks like:

```
0.000: (move robot1 assembly_zone steering_wheels_zone)  [5.000]
5.001: (pickup robot2 body_car_1 body_car_zone)  [5.000]
5.001: (pickup robot1 steering_wheel_1 steering_wheels_zone)  [5.000]
10.001: (move robot2 body_car_zone assembly_zone)  [5.000]
10.001: (move robot1 steering_wheels_zone assembly_zone)  [5.000]
15.002: (dropoff robot2 body_car_1 assembly_zone)  [5.000]
15.002: (dropoff robot1 steering_wheel_1 assembly_zone)  [5.000]     <-- assembly begins after this step
20.003: (move robot2 assembly_zone wheels_zone)  [5.000]
25.004: (pickup robot2 wheel_1 wheels_zone)  [5.000]
30.004: (move robot2 wheels_zone assembly_zone)  [5.000]
35.005: (dropoff robot2 wheel_1 assembly_zone)  [5.000]
40.006: (assemble robot1 assembly_zone wheel_1 body_car_1 steering_wheel_1 car_1)  [5.000]
```
This is not an issue when using the `foxy-devel` release. Seems like no matter how the plan might look via POPF extensive testing on how it behaves in the skill layer is required.

We will have a working example in the very near future PR'ed to the example repo to demonstrate what we're seeing and how we're approach the multi-bot problem. Hopefully this will clarify what we're seeing.

fmrico, 2021-02-03:
Hi @sethhoward 

May I have the complete pddl, including dropoff, to do some tests?

sethhoward, 2021-02-04:
Absolutely, I realized we haven't tried this domain with a solitary robot on either foxy or main branch.

```(define (domain factory)
(:requirements :strips :typing :adl :fluents :durative-actions)

;; Types ;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;
(:types
robot
zone
piece
car
);; end Types ;;;;;;;;;;;;;;;;;;;;;;;;;

;; Predicates ;;;;;;;;;;;;;;;;;;;;;;;;;
(:predicates

(robot_at ?r - robot ?z - zone)
(piece_at ?p - piece ?z - zone)

(piece_is_wheel ?p - piece)
(piece_is_body_car ?p - piece)
(piece_is_steering_wheel ?p - piece)

(piece_not_used ?p - piece)

(is_assembly_zone ?z - zone)
(carry ?r - robot ?p - piece)

(car_assembled ?c - car)

);; end Predicates ;;;;;;;;;;;;;;;;;;;;
;; Functions ;;;;;;;;;;;;;;;;;;;;;;;;;
(:functions

);; end Functions ;;;;;;;;;;;;;;;;;;;;
;; Actions ;;;;;;;;;;;;;;;;;;;;;;;;;;;;
(:durative-action move
    :parameters (?r - robot ?z1 ?z2 - zone)
    :duration ( = ?duration 5)
    :condition (and
        (at start(robot_at ?r ?z1)))
    :effect (and
        (at start(not(robot_at ?r ?z1)))
        (at end(robot_at ?r ?z2))
    )
)

(:durative-action pickup
    :parameters (?r - robot ?p - piece ?z - zone)
    :duration ( = ?duration 5)
    :condition (and
        (at start(robot_at ?r ?z))
        (over all (robot_at ?r ?z))
        
        (at start(piece_at ?p ?z))
    )
    :effect (and
        (at end(carry ?r ?p))
        (at end (not (piece_at ?p ?z)))
        (at start (not(piece_not_used ?p)))
        
        ; prevents a double move where both bots attempt to fetch the wheel even though only on can.
        ; also, Plansys2 skill layer will fail to execute the pick and place actions.
        (at start(robot_at ?r ?z))
    )
)

(:durative-action dropoff
    :parameters (?r - robot ?p - piece ?z - zone)
    :duration ( = ?duration 5)
    :condition (and
        (at start(robot_at ?r ?z))
        (over all (robot_at ?r ?z))
        
        (at start(carry ?r ?p))
    )
    :effect (and
        (at end(piece_not_used ?p))
        (at end(not(carry ?r ?p)))
        (at end(piece_at ?p ?z))
        
       ; Plansys2 skill layer fails without this even though the plan looks solid
        (at start (robot_at ?r ?z))
    )
)

(:durative-action assemble
    :parameters (?r - robot ?z - zone ?p1 ?p2 ?p3 - piece ?c - car)
    :duration ( = ?duration 5)
    :condition (and
        (at start(is_assembly_zone ?z))
        (at start(robot_at ?r ?z))
        (over all (robot_at ?r ?z))

        (at start(piece_at ?p1 ?z))
        (at start(piece_at ?p2 ?z))
        (at start(piece_at ?p3 ?z))

        (at start(piece_not_used ?p1))
        (at start(piece_not_used ?p2))
        (at start(piece_not_used ?p3))

        (at start(piece_is_wheel ?p1))
        (at start(piece_is_body_car ?p2))
        (at start(piece_is_steering_wheel ?p3))
    )
    :effect (and
        (at start(not(piece_not_used ?p1)))
        (at start(not(piece_not_used ?p2)))
        (at start(not(piece_not_used ?p3)))
        (at end(car_assembled ?c))

    )
)

);; end Domain ;;;;;;;;;;;;;;;;;;;;;;;;```

sethhoward, 2021-02-04:
@fmrico 

Foxy release with the domain above fails to run the gripper tasks with one robot unless the `(at start(robot_at ?r ?z))` guard is included in `pickup`. 

The Main branch running the same domain works fine using the above domain and also removing `(at start(robot_at ?r ?z))`

```  
        ...

        (at start (not(piece_not_used ?p)))
        
        ; prevents a double move where both bots attempt to fetch the wheel even though only on can.
        ; also, Plansys2 skill layer will fail to execute the pick and place actions.
       ; (at start(robot_at ?r ?z))

       ...
```
It also successfully complete the mission with one robot. The issue of early assembly appears only when running multiple actors:

```
0.000: (move robot1 assembly_zone steering_wheels_zone)  [5.000]
5.001: (pickup robot2 body_car_1 body_car_zone)  [5.000]
5.001: (pickup robot1 steering_wheel_1 steering_wheels_zone)  [5.000]
10.001: (move robot2 body_car_zone assembly_zone)  [5.000]
10.001: (move robot1 steering_wheels_zone assembly_zone)  [5.000]
15.002: (dropoff robot2 body_car_1 assembly_zone)  [5.000]
15.002: (dropoff robot1 steering_wheel_1 assembly_zone)  [5.000]     <-- assembly begins after this step
20.003: (move robot2 assembly_zone wheels_zone)  [5.000]
25.004: (pickup robot2 wheel_1 wheels_zone)  [5.000]
30.004: (move robot2 wheels_zone assembly_zone)  [5.000]
35.005: (dropoff robot2 wheel_1 assembly_zone)  [5.000]
40.006: (assemble robot1 assembly_zone wheel_1 body_car_1 steering_wheel_1 car_1)  [5.000]
```

```
(define (problem factory_1) (:domain factory)
(:objects 
    robot1 - robot
    robot2 - robot
    wheels_zone - zone
    steering_wheels_zone - zone
    body_car_zone - zone
    assembly_zone - zone
    wheel_1 - piece
    body_car_1 - piece
    steering_wheel_1 - piece
    car_1 - car
)

(:init
    (robot_at robot1 assembly_zone)
    (robot_at robot2 assembly_zone)
    (is_assembly_zone assembly_zone)
    (piece_at wheel_1 wheels_zone)
    (piece_at body_car_1 body_car_zone)
    (piece_at steering_wheel_1 steering_wheels_zone)
    (piece_is_wheel wheel_1)
    (piece_is_body_car body_car_1)
    (piece_is_steering_wheel steering_wheel_1)
    (piece_not_used wheel_1)
    (piece_not_used body_car_1)
    (piece_not_used steering_wheel_1)
)

(:goal (and
    (car_assembled car_1)
))
```

Here are the BT definitions:

Pickup
```
<root main_tree_to_execute = "MainTree" >
    <BehaviorTree ID="MainTree">
       <Sequence name="root_sequence">
           <OpenGripper    name="open_gripper"/>
           <ApproachObject name="approach_object"/>
           <CloseGripper   name="close_gripper"/>
       </Sequence>
    </BehaviorTree>
</root>
```

Drop off

```<root main_tree_to_execute = "MainTree" >
    <BehaviorTree ID="MainTree">
       <Sequence name="root_sequence">
           <OpenGripper    name="open_gripper"/>
       </Sequence>
    </BehaviorTree>
</root>
```

fmrico, 2021-02-28:
@sethhoward  I think I found the problem. Working on it...

fmrico, 2021-03-13:
Hi @sethhoward 

This Bug should be already fixed with the last refactors? Could you check it, please.

sethhoward, 2021-03-25:
Sorry for the delay. I checked out the latest Foxy-devel branch. Using the same project as above as well as another project running in a different environment, map and doing a simple pick/place (no assembly).

I'm unsure if the refactors have fixed the issues because we're failing after the pickup! 

```[bt_action_node-30] CloseGripper tick 4
[bt_action_node-30] CloseGripper tick 5
[bt_action_node-30] OpenGripper halt
[bt_action_node-30] ApproachObject halt
[bt_action_node-30] CloseGripper halt
[bt_action_node-29] [WARN] [1616703436.362754649] [robot.move_robot]: current and specialized arguments lenght doesn't match 3 1
[bt_action_node-29] [INFO] [1616703436.367674280] [robot.move_robot]: Waiting for "navigate_to_pose" action server
[bt_action_node-29] [INFO] [1616703436.368123553] [robot.move_robot]: "move" BtActionNode initialized
[bt_action_node-29] [ERROR] [1616703436.372999917] []: Caught exception in callback for transition 13
[bt_action_node-29] [ERROR] [1616703436.373020478] []: Original error: parameter 'waypoints' has already been declared
[bt_action_node-29] [WARN] [1616703436.373034937] []: Error occurred while doing error handling.```

Thoughts? Anything else I can provide?

sethhoward, 2021-03-29:
Aside from my comments (which are more of a usability/client issue). This bug has been fixed. Thanks!

fmrico, 2021-03-29:
Thanks for testing, @sethhoward

closed 2021-03-29 by fmrico