PlanSys2/ros2_planning_system #80: Action node implementig several actions.

teyssieuman, 2021-01-26:
In order to shorten the problem description, it can be a useful trick to make several quite similar actions. For example:

* travel_road_direct     road a b
* travel_road_reverse  road a b

Is this a way to create an action node that implements two actions ?

fmrico, 2021-01-26:
You can run/instantiate two nodes of the same action. There is no problem with that. In plansys2_bt_example I have this in the launcher to execute several actions in parallel:

```
    # Specify the actions
    move_1_cmd = Node(
        package='plansys2_bt_actions',
        executable='bt_action_node',
        name='move_1',
        namespace=namespace,
        output='screen',
        parameters=[
          example_dir + '/config/params.yaml',
          {
            'action_name': 'move',
            'bt_xml_file': example_dir + '/behavior_trees_xml/move.xml'
          }
        ])

    move_2_cmd = Node(
        package='plansys2_bt_actions',
        executable='bt_action_node',
        name='move_2',
        namespace=namespace,
        output='screen',
        parameters=[
          example_dir + '/config/params.yaml',
          {
            'action_name': 'move',
            'bt_xml_file': example_dir + '/behavior_trees_xml/move.xml'
          }
        ])

    move_3_cmd = Node(
        package='plansys2_bt_actions',
        executable='bt_action_node',
        name='move_3',
        namespace=namespace,
        output='screen',
        parameters=[
          example_dir + '/config/params.yaml',
          {
            'action_name': 'move',
            'bt_xml_file': example_dir + '/behavior_trees_xml/move.xml'
          }
        ])
```

If you have three robots, you can use the `specialized_arguments` parameter to make that a robot only performs actions that contain its id as part of the action. For example, if you set `specialized_arguments` for action `move`  to `robot1`. The `robot1` only executes the action `move that contains `robot1` among any action parameters.

fmrico, 2021-01-26:
Reading again your question, I think I wasn't precise. 

No. An action performer only executes one action.

fmrico, 2021-02-03:
Hi @teyssieuman . Can we close this issue? Do you think we should take an action about this?

teyssieuman, 2021-02-03:
Hi @fmrico I managed to reorganize my nodes to implement another solution. So I think that this issue can be close.

closed 2021-02-03 by teyssieuman