PlanSys2/ros2_planning_system #81: Add support for numeric conditions and effects

jjzapf, 2021-01-29:
The pddl parser is able to parse a number of constructs that are not supported by the rest of the ROS2 Planning System. In particular, the system cannot be used to execute actions involving numeric conditions and effects.

To allow for the use of numeric conditions and effects and make it easier to add support for additional constructs in the future, I believe we should modify the way constructs are passed between the various ROS2 Planning System modules. Currently, PDDLPrint() functions in the pddl parser are called to return strings to the domain expert. These strings are then converted to predicate trees by the domain expert. This intermediary string conversion seems unnecessary. The code encapsulated by the Types.hpp/cpp files in the domain expert should be moved to the pddl parser in new files called Tree.h/cpp. We should then add PDDLTree() functions to the various pddl construct classes in the pddl parser. These functions would directly compute the predicate trees from the pddl parser objects rather than use the intermediary string conversions.

This issue will be used to add support for numeric conditions and effects using the proposed strategy described above. These changes have already been made in my forked version of the repository.

fmrico, 2021-02-03:
Hi @jjzapf 

I think that you resolved this issue in your PR that I have just merged, didn't you? Can we close this issue?

jjzapf, 2021-02-03:
Yes. Thanks for merging the changes!

closed 2021-02-03 by jjzapf