PlanSys2/ros2_planning_system #86: Visualize the execution of the actions implemented as behavior trees

xydesa, 2021-02-05:
#85 recently added support to the plansys2_executor/ExecutorNode to visualize the executing BehaviorTree for the current plan using the [Groot](https://github.com/BehaviorTree/Groot) tool.

This visualization may also be useful for BehaviorTree actions that are themselves implemented as BehaviorTrees. Care would have to be taken to make sure that none of the ZeroMQ publishers that connect to Groot use the same ports.

xydesa, 2021-02-05:
@fmrico does this summarize what you were talking about in #85?

Also, it seems to me that the [`BTActionNode`](https://github.com/IntelligentRoboticsLabs/ros2_planning_system/blob/master/plansys2_bt_actions/include/plansys2_bt_actions/BTActionNode.hpp) just implements a base class for BehaviorTree action nodes to be used with plansys2. Is that correct? If this is correct it doesn't seem like the right place for this type of visualization as not all of the actions nodes used in a behavior tree need to be behavior trees themselves.

Were you thinking of adding this support to `BtActionNode` or [`BtAction`](https://github.com/IntelligentRoboticsLabs/ros2_planning_system/blob/master/plansys2_bt_actions/src/plansys2_bt_actions/BTAction.cpp)?
It seems like it would be pretty straightforward to add this visualization to `BtAction` as it contains a BehaviorTree object that could be visualized by Groot.

fmrico, 2021-02-06:
`bt_action_node` is a node that implamentes an action performer that executes a Behavior Tree. If you have a look [here](https://github.com/IntelligentRoboticsLabs/ros2_planning_system_examples/tree/master/plansys2_bt_example), the launchers contains this:

```
    move_1_cmd = Node(
        package='plansys2_bt_actions',
        executable='bt_action_node',
        name='move_1',
        namespace=namespace,
        output='screen',
        parameters=[
          example_dir + '/config/params.yaml',
          {
            'action_name': 'move',
            'bt_xml_file': example_dir + '/behavior_trees_xml/move.xml'
          }
        ])
  
   [...]

    transport_1_cmd = Node(
        package='plansys2_bt_actions',
        executable='bt_action_node',
        name='transport_1',
        namespace=namespace,
        output='screen',
        parameters=[
          example_dir + '/config/params.yaml',
          {
            'action_name': 'transport',
            'bt_xml_file': example_dir + '/behavior_trees_xml/transport.xml'
          }
        ]) 
```

The executable is the same, but you specify the XML and the `action_name` by parameters. It would be cool if these actions can be visualized separately.

PR #91 (2021-02-21): Add Groot ZeroMQ-based visualization of the BehaviorTree in each BTActionNode.
Fixes #86 

*Testing*
* Basing the test off of the bt example in the ros2_planning_system_examples repository.
  * https://github.com/IntelligentRoboticsLabs/ros2_planning_system_examples/tree/master/plansys2_bt_example
* Create a workspace from this repos file: 
[bt-action-visualization.repos.txt](https://github.com/IntelligentRoboticsLabs/ros2_planning_system/files/5991987/bt-action-visualization.repos.txt)
* Terminal 1: `ros2 launch nav2_bringup tb3_simulation_launch.py`
* Terminal 2: `ros2 launch plansys2_bt_exa [... cut]
closed 2021-02-21 by fmrico