PlanSys2/ros2_planning_system #87: `and` not supported in `durative-action` conditions or effects.

teyssieuman, 2021-02-08:
Considering the two following durative actions:
```pddl
(:durative-action A
  :parameters ( )
  :duration (= ?duration 1)
  :conditions ( and
    (at start <predicate_1>)
    (at start <predicate_2>)
  )
  :effect (and
    (at end <predicate_3>)
    (at end <predicate_4>)
) 

(:durative-action B
  :parameters ( )
  :duration (= ?duration 1)
  :conditions ( and
    (at start (and ( <predicate_1> <predicate_2> ) ) )
  )
  :effect (and
    (at end (and ( <predicate_3> <predicate_4> ) ) )
) 
```

The durative actions `A` and `B` are accepted by POPF witch find a plan.

But the `B` action is not working correctly withe `plansys2`  :
- In `BTBuilder::get_base` the `parser::pddl::tree::AND` node is not supported in the `FUNCTION` case.

fmrico, 2021-03-13:
Hi @teyssieuman 

Does this bug still exist after the last refactors?

fmrico, 2021-05-18:
Hi @teyssieuman 

I am closing this issue. Pease, reopen it if this effect appears again in the last version.

Thanks!!

closed 2021-05-18 by fmrico