PlanSys2/ros2_planning_system #89: Cancelling a plan

rossj13, 2021-02-16:
I can't seem to see how I would cancel/ clear the problem client and executor client part way through executing a plan so I could make a new plan to execute. Am I missing something or is that something that can be added in future instances.

fmrico, 2021-02-16:
Hi @rossj13 

Just this week I am working on this. Stay tuned.

tdenewiler, 2021-02-19:
I'm having similar issues at the moment. I'd like to track any changes if it would help. I can build your development branch and test with my own code to provide feedback.

fmrico, 2021-02-19:
Hi @tdenewiler 

I have been working on this all week. I have fixed it, but I am adding a lot of tests to be sure that everything will be working. A think that this weekend I will do a PR. Maybe you could test/review it before merging when I create it.

tdenewiler, 2021-02-19:
I'll do my best to monitor your PRs and will let you know if I am able to test it this weekend. Thanks!

fmrico, 2021-03-07:
Hi @rossj13 @tdenewiler 

I am closing this, as canceling plans are available from #92 

Please, open it again if you find any issue

closed 2021-03-07 by fmrico