PlanSys2/ros2_planning_system #112: Guidance on Solver Arguments

kogut, 2021-04-13:
I'd like to add the ability to provide arguments to the popf solver, which has many significant arguments. 

The first way that occurred to me to implement it is via a change to the plugin API, e.g. adding an optional string argument to GetPlan().   But I wanted to check here, as that would be a breaking change to the API.   I'm happy to do the work to PR to this repo and the 3 external plugins that I know of.  

Or if there's a better way, or you'd prefer not to change the plugin API at all, let me know.

fmrico, 2021-04-14:
Hi @kogut 

I would use a parameter to specify how to call the solver, directly readen from the popf plugin.

I would prefer not to change the API.

kogut, 2021-04-14:
@fmrico, yes, that was my 2nd choice.    Will proceed that way.    Thank you for the feedback.

closed 2021-04-14 by kogut