PlanSys2/ros2_planning_system #113: Plansys2_node dies

jginesclavero, 2021-04-14:
Hi @fmrico !
Sometimes plansys2_node dies when an action finishes. Here you have the output.

```
[plansys2_node-15] [PublisherZMQ] Server quitting.
[plansys2_node-15] [PublisherZMQ] just died. Exeption Context was terminated
[plansys2_node-15] [PublisherZMQ] Publisher quitting.
[plansys2_node-15] [PublisherZMQ] just died. Exeption Context was terminated
[plansys2_node-15] pure virtual method called
[plansys2_node-15] terminate called without an active exception
[ERROR] [plansys2_node-15]: process has died [pid 192872, exit code -6, cmd '/home/jgines/robocup_ws/install
/plansys2_bringup/lib/plansys2_bringup/plansys2_node --ros-args -r __ns:=/ --params-file /tmp/launch_params_f6d2888d 
--params-file /home/jgines/robocup_ws/install/plansys2_bringup/share/plansys2_bringup/params/plansys2_params.yaml'].
```

fmrico, 2021-04-18:
Hi @jginesclavero 

May I have more details? 
* Domain? 
* Action performers?
* Is it happening the same with "distributed" launcher? I guess you are using monolithic...

Thanks

fmrico, 2021-05-16:
Probably this is fixed in #122

closed 2021-05-16 by fmrico