PlanSys2/ros2_planning_system #120: if

lingxd, 2021-05-06:


fmrico, 2021-05-16:
Well, this is actually a good question, @lingxd .

The examples are always pointing to Plansys2 `master`, but I should probably tag examples with ros-foxy. Anyway, you can see when the `plansys2-foxy` branch was created and check out the version of the example corresponding to that time.

Let me know if that works for you.

Best

lingxd, 2021-05-17:
this is my build log
```shell
hokori@hokori:WorkSpaces/ros2_planning_system_examples $ colcon build --symlink-install

Starting >>> plansys2_bt_example
Starting >>> plansys2_cascade_example
Starting >>> plansys2_multidomain_example
Starting >>> plansys2_patrol_navigation_example
Starting >>> plansys2_simple_example
--- stderr: plansys2_cascade_example                                                                                                                                                                                                                 
/home/hokori/Documents/WorkSpaces/ros2_planning_system_examples/src/plansys2_cascade_example/src/move_action_node.cpp:21:10: fatal error: plansys2_msgs/action/execute_action.hpp: 没有那个文件或目录
   21 | #include "plansys2_msgs/action/execute_action.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/move_action_node.dir/build.make:63：CMakeFiles/move_action_node.dir/src/move_action_node.cpp.o] 错误 1
make[1]: *** [CMakeFiles/Makefile2:165：CMakeFiles/move_action_node.dir/all] 错误 2
make[1]: *** 正在等待未完成的任务....
/home/hokori/Documents/WorkSpaces/ros2_planning_system_examples/src/plansys2_cascade_example/src/charge_action_node.cpp:18:10: fatal error: plansys2_msgs/action/execute_action.hpp: 没有那个文件或目录
   18 | #include "plansys2_msgs/action/execute_action.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/charge_action_node.dir/build.make:63：CMakeFiles/charge_action_node.dir/src/charge_action_node.cpp.o] 错误 1
make[1]: *** [CMakeFiles/Makefile2:111：CMakeFiles/charge_action_node.dir/all] 错误 2
/home/hokori/Documents/WorkSpaces/ros2_planning_system_examples/src/plansys2_cascade_example/src/ask_charge_action_node.cpp:18:10: fatal error: plansys2_msgs/action/execute_action.hpp: 没有那个文件或目录
   18 | #include "plansys2_msgs/action/execute_action.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/ask_charge_action_node.dir/build.make:63：CMakeFiles/ask_charge_action_node.dir/src/ask_charge_action_node.cpp.o] 错误 1
make[1]: *** [CMakeFiles/Makefile2:138：CMakeFiles/ask_charge_action_node.dir/all] 错误 2
make: *** [Makefile:141：all] 错误 2
---
Failed   <<< plansys2_cascade_example [0.25s, exited with code 2]
Aborted  <<< plansys2_patrol_navigation_example [0.24s]
Aborted  <<< plansys2_multidomain_example [0.25s]
Aborted  <<< plansys2_simple_example [0.24s]
Aborted  <<< plansys2_bt_example [3.63s]                               

Summary: 0 packages finished [3.74s]
  1 package failed: plansys2_cascade_example
  4 packages aborted: plansys2_bt_example plansys2_multidomain_example plansys2_patrol_navigation_example plansys2_simple_example
  5 packages had stderr output: plansys2_bt_example plansys2_cascade_example plansys2_multidomain_example plansys2_patrol_navigation_example plansys2_simple_example
```

and ros-foxy-plansys2-msgs is installed.

lingxd, 2021-05-17:
```output
git branch -vv

* foxy-devel 90cd31e [origin/foxy-devel] Minor fix
```

fmrico, 2021-05-17:
Hi!! Did you clone [PlanSys2](https://github.com/IntelligentRoboticsLabs/ros2_planning_system) in the src in your workspace?

It seems like only the examples are in the workspace...

lingxd, 2021-05-17:
yeah, I followed this tutorial.
https://intelligentroboticslab.gsyc.urjc.es/ros2_planning_system.github.io/getting_started/index.html

lingxd, 2021-05-17:
https://github.com/IntelligentRoboticsLabs/ros2_planning_system_examples/issues/18

fmrico, 2021-05-17:
> git branch -vv
> 
> * foxy-devel 90cd31e [origin/foxy-devel] Minor fix

Use `master` branch, please. `foxy-devel` is only for packaging.

lingxd, 2021-05-17:
thanks!
I compiled it successfully, but how does the example run successfully?

lingxd, 2021-05-17:
when i run `run`

```output
> run
[INFO] [1621242528.194967859] [terminal]: Already executing a plan
Execution could not start 

```

Is that a success?

closed 2021-05-18 by fmrico