PlanSys2/ros2_planning_system #125: Crashing when running plan

devis12, 2021-06-01:
Good afternoon,
is it possible to have further insights into the following problem which I do not seem to understand on my own.
I've written the following pddl domain:

`;; domain file: turtle-eat-domain.pddl

(define (domain turtle-eat-domain)

    (:requirements :strips :typing :adl :fluents :durative-actions)

    ;; Types ;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;
    (:types
        turtle
    );; end Types ;;;;;;;;;;;;;;;;;;;;;;;;;

    (:predicates 
        (control ?t - turtle)
        (targetFree ?t - turtle)
        (currentTarget ?t - turtle)
        (alive ?t - turtle)
        (eaten ?t - turtle)
    )

    (:functions
        (turtlesEaten ?t - turtle)
        (distance ?t1 ?t2 - turtle)
    )

    (:durative-action acquiretarget
        :parameters (?t1 ?t2 - turtle)
        :duration (= ?duration 1)
        :condition (and
            (at start (targetFree ?t1))
            (over all (control ?t1))
            (over all (alive ?t2))
        )
        :effect (and
            (at start (currentTarget ?t2))
            (at start (not(targetFree ?t1)))
        )
    )

    (:durative-action movetoward
        :parameters (?t1 ?t2 - turtle)
        :duration (= ?duration 4)
        :condition (and
            (at start (> (distance ?t1 ?t2) 0))
            (over all (control ?t1))
            (over all (currentTarget ?t2))
            (over all (alive ?t2))
        )
        :effect (and
            (at end (assign (distance ?t1 ?t2) 0))
        )
    )

    (:durative-action eating
        :parameters (?t1 ?t2 - turtle)
        :duration (= ?duration 2)
        :condition (and
            (over all (control ?t1))
            (over all (currentTarget ?t2))
            (over all (alive ?t2))
            (over all (< (distance ?t1 ?t2) 1))
        )
        :effect (and
            (at end (not(alive ?t2)))
            (at end (eaten ?t2))
            (at end (not (currentTarget ?t2)))
            (at end (targetFree ?t1))
            (at end (increase (turtlesEaten ?t1) 1))
        )
    )

)`

I've prepared three nodes which are supposed to run the actions, but up to now, they act just as in your first example (increase the progress regularly and print it out without doing anything else).

With the domain loaded, I open the plansys2 terminal and I give the following commands to instantiate a problem:
`set instance turtle1 turtle
set instance turtle2 turtle
set instance turtle3 turtle
set instance turtle4 turtle
set instance turtle5 turtle

set function (turtlesEaten turtle1 0)
set function (distance turtle1 turtle2 3)
set function (distance turtle1 turtle3 3)
set function (distance turtle1 turtle4 3)
set function (distance turtle1 turtle5 3)

set predicate (control turtle1)
set predicate (targetFree turtle1)
set predicate (alive turtle1)
set predicate (alive turtle2)
set predicate (alive turtle3)
set predicate (alive turtle4)
set predicate (alive turtle5)

set goal (and (eaten turtle2) (eaten turtle3) (eaten turtle4) (eaten turtle5))`

I prompt `get plan` and the plan is correctly computed:
plan: 
0	(acquiretarget turtle1 turtle2)	1
0.001	(movetoward turtle1 turtle2)	4
4.002	(eating turtle1 turtle2)	2
6.003	(acquiretarget turtle1 turtle3)	1
6.003	(movetoward turtle1 turtle3)	4
10.004	(eating turtle1 turtle3)	2
12.005	(acquiretarget turtle1 turtle4)	1
12.005	(movetoward turtle1 turtle4)	4
16.006	(eating turtle1 turtle4)	2
18.007	(acquiretarget turtle1 turtle5)	1
18.007	(movetoward turtle1 turtle5)	4
22.008	(eating turtle1 turtle5)	2

However, plansys2 seems not to be able to run the plan, the logs in the terminal are the following:
[plansys2_node-1] [INFO] [1622553543.485785116] [executor]: Action acquiretarget timeout percentage -1.000000
[plansys2_node-1] [INFO] [1622553543.489139589] [executor]: Action movetoward timeout percentage -1.000000
[plansys2_node-1] [INFO] [1622553543.492199201] [executor]: Action eating timeout percentage -1.000000
[plansys2_node-1] [INFO] [1622553543.495399870] [executor]: Action acquiretarget timeout percentage -1.000000
[plansys2_node-1] [INFO] [1622553543.498682337] [executor]: Action movetoward timeout percentage -1.000000
[plansys2_node-1] [INFO] [1622553543.501892589] [executor]: Action eating timeout percentage -1.000000
[plansys2_node-1] [INFO] [1622553543.505093113] [executor]: Action acquiretarget timeout percentage -1.000000
[plansys2_node-1] [INFO] [1622553543.508346455] [executor]: Action movetoward timeout percentage -1.000000
[plansys2_node-1] [INFO] [1622553543.511911271] [executor]: Action eating timeout percentage -1.000000
[plansys2_node-1] [INFO] [1622553543.516298207] [executor]: Action acquiretarget timeout percentage -1.000000
[plansys2_node-1] [INFO] [1622553543.519800856] [executor]: Action movetoward timeout percentage -1.000000
[plansys2_node-1] [INFO] [1622553543.523239628] [executor]: Action eating timeout percentage -1.000000
**[plansys2_node-1] ===> [(targetfree turtle1)]
[plansys2_node-1] plansys2_node: /home/devis/ros2_ws/src/ros2_planning_system/plansys2_executor/src/plansys2_executor/BTBuilder.cpp:521: plansys2::Graph::Ptr plansys2::BTBuilder::get_graph(const Plan&): Assertion `at_start_requirements.empty()' failed.
[ERROR] [plansys2_node-1]: process has died [pid 22317, exit code -6, cmd '/home/devis/ros2_ws/install/plansys2_bringup/lib/plansys2_bringup/plansys2_node --ros-args -r __ns:=/ --params-file /tmp/launch_params_hyoy7vbc --params-file /home/devis/ros2_ws/install/plansys2_bringup/share/plansys2_bringup/params/plansys2_params.yaml'].**

Can you give me an hint at what's going on?
If you need the entire pkg and/or more info, let me know.
Thanks

P.S. I've tried removing functions from the domain (since in another working example which I did with the hanoi domain and just  a single durative action, without any function definition, it worked), but it doesn't solve the problem. I'm clueless.

roveri-marco, 2021-06-01:
Hi all, the problem is related to the fact that although the domain uses `targetFree` (see below)

```
(:durative-action` acquiretarget
    :parameters (?t1 ?t2 - turtle)
    :duration (= ?duration 1)
    :condition (and
        (at start (targetFree ?t1))
        (over all (control ?t1))
        (over all (alive ?t2))
    )
    :effect (and
        (at start (currentTarget ?t2))
        (at start (not(targetFree ?t1)))
    )
)
```

In plansys2 the action is stored as "targetfree"

```
(:durative-action acquiretarget
        :parameters (?t1 ?t2 - turtle)
        :duration (= ?duration 1)
        :condition (and
            (at start (targetfree ?t1))
            (over all (control ?t1))
            (over all (alive ?t2))
        )
        :effect (and
            (at start (currenttarget ?t2))
            (at start (not(targetfree ?t1)))
        )
    )
```

In the problem the problem is specified as

`set predicate (targetFree turtle1)`

And thus, the executor when checking the precondition fails since in the knowledge base it does not find the predicate.

I believe the same conversion approach should be applied in both cases.

fmrico, 2021-06-02:
Hi,

Thanks for investigating this bug. I will try to reproduce it by myself and see how to fix it. We should be transforming all to lowercase to avoid these kid of things...

Best

fmrico, 2021-06-02:
Fixed in #128, but waiting for @jjzapf 's opinion

closed 2021-09-01 by fmrico