PlanSys2/ros2_planning_system #140: Running of Planner Plugins Limited to 15 Seconds, and Provide No Feedback

kogut, 2021-06-15:
Currently, if the underlying plan solver takes longer than 15 seconds to return with a plan, the planner client silently returns without a plan:

https://github.com/IntelligentRoboticsLabs/ros2_planning_system/blob/master/plansys2_planner/src/plansys2_planner/PlannerClient.cpp#L54

Also, no feedback is provided about how long the underlying planner is taking.

These are fine for relatively simple plans.  But can be problematic for plans with large search spaces.

To better accommodate more complex planning tasks, I propose:

1) Making the timeout configurable (ROS parameter).  Default would remain 15 seconds.
2) Converting the PlannerNode interface from a service to an action.  The feedback would - at least initially - be the percentage of the timeout period.  

This would accommodate plans that take an arbitrary length of time to compute.

If this sounds reasonable, I will undertake this work.

fmrico, 2021-06-16:
Hi @kogut 

I think it is perfectly reasonable. My concern is increasing the complexity in codes like https://github.com/IntelligentRoboticsLabs/ros2_planning_system_examples/blob/3b03903ad3bf8089a88ff767dd99f8c9cebbc165/plansys2_patrol_navigation_example/src/patrolling_controller_node.cpp#L78.

This complexity should be managed inside the PlannerClient. Maybe we could maintain the current blocking call, and add another non-blocking call (using std::futures?) so the users could have both alternatives, and maintain backward compatibility.

Regarding timeout, maybe it could be interesting to have a default timeout, set in parameters (with a default value in case of not being set). We could also add a last parameter `timeout` to the `getPlan` method, using this default value in case of not set.

What do you think? Any other contributions to this discussion, @teyssieuman @xydesa @jjzapf @jginesclavero ?

xydesa, 2021-06-16:
Given that solving a plan can take a varying amount of time, I think using an action makes sense for the underlying connection between the plansys2_planner/PlannerClient and the plansys2_planner/PlannerNode. Actions were built for longer running tasks like this: https://docs.ros.org/en/foxy/Tutorials/Understanding-ROS2-Actions.html#background
The best way to do this may involve changing the plansys2::PlanSolverBase interface to add feedback from the planner (if available).

I also like the idea of having both a blocking and non-blocking function call on the plansys2_planner/PlannerClient so developers could use whichever one they prefer. The non-blocking call could also provide feedback on progress. I'd be ok with just keeping the blocking function call for now and leaving a non-blocking call as a future enhancement.

I think a default timeout should definitely be set and a parameter provided to override that. I also like the idea of adding a timeout parameter to the plansys2_planner/PlannerClient::getPlan function, this would let other developers choose how long they'd like to wait.

fmrico, 2022-07-09:
Closing... reopen if you want to continue with this

closed 2022-07-09 by fmrico