PlanSys2/ros2_planning_system #143: executor client: Plan Failed

tammerb, 2021-06-26:
I've tried running the full BT example in the tutorials with `move` `transport` and `assemble` actions but am unable to run the full plan. Here is my knowledge initialization:
`set instance r2d2 robot`
`set instance wheels_zone zone`
`set instance steering_wheels_zone zone`
`set instance body_car_zone zone`
`set instance assembly_zone zone`
`set instance recharge_zone zone`
`set instance wheel_1 piece`
`set instance body_car_1 piece`
`set instance steering_wheel_1 piece`
`set instance car_1 car`
`set predicate (robot_at r2d2 body_car_zone)`
`set predicate (is_assembly_zone assembly_zone)`
`set predicate (piece_at wheel_1 wheels_zone)`
`set predicate (piece_at body_car_1 body_car_zone)`
`set predicate (piece_at steering_wheel_1 steering_wheels_zone)`
`set predicate (is_recharge_zone recharge_zone)`
`set predicate (piece_is_wheel wheel_1)`
`set predicate (piece_is_body_car body_car_1)`
`set predicate (piece_is_steering_wheel steering_wheel_1)`
`set predicate (piece_not_used wheel_1)`
`set predicate (piece_not_used body_car_1)`
`set predicate (piece_not_used steering_wheel_1)`
`set predicate (robot_available r2d2)`
`set predicate (battery_full r2d2)`
`set goal (and(car_assembled car_1))`

The following plan is generated:
`plan:` 
`0       (transport r2d2 body_car_1 body_car_zone assembly_zone)    5`
`5.001   (move r2d2 assembly_zone steering_wheels_zone)     5`
`10.002  (transport r2d2 steering_wheel_1 steering_wheels_zone assembly_zone)       5`
`15.003  (move r2d2 assembly_zone wheels_zone)   5`
`20.004  (transport r2d2 wheel_1 wheels_zone assembly_zone) 5`
`25.005  (assemble r2d2 assembly_zone wheel_1 body_car_1 steering_wheel_1 car_1)    5`

However when I run, the plansys2_bt_example node shows the following output:
`[bt_action_node-5] [INFO] [1624737309.265907883] [transport_1]: Waiting for "navigate_to_pose" action server`
`[bt_action_node-5] [INFO] [1624737309.267119276] [transport_1]: "move" BtActionNode initialized`
`[bt_action_node-5] OpenGripper tick 0`
`[bt_action_node-5] OpenGripper tick 1`
`[bt_action_node-5] OpenGripper tick 2`
`[bt_action_node-5] OpenGripper tick 3`
`[bt_action_node-5] OpenGripper tick 4`
`[bt_action_node-5] OpenGripper tick 5`
`[bt_action_node-5] ApproachObject tick 0`
`[bt_action_node-5] ApproachObject tick 1`
`[bt_action_node-5] ApproachObject tick 2`
`[bt_action_node-5] ApproachObject tick 3`
`[bt_action_node-5] ApproachObject tick 4`
`[bt_action_node-5] ApproachObject tick 5`
`[bt_action_node-5] CloseGripper tick 0`
`[bt_action_node-5] CloseGripper tick 1`
`[bt_action_node-5] CloseGripper tick 2`
`[bt_action_node-5] CloseGripper tick 3`
`[bt_action_node-5] CloseGripper tick 4`
`[bt_action_node-5] CloseGripper tick 5`
`[bt_action_node-5] [INFO] [1624737312.295797407] [transport_1]: Waiting for "navigate_to_pose" action server`
`[bt_action_node-5] [PublisherZMQ] Server quitting.`
`[bt_action_node-5] [PublisherZMQ] just died. Exeption Context was terminated`
`[bt_action_node-5] [PublisherZMQ] Publisher quitting.`
`[bt_action_node-5] [PublisherZMQ] just died. Exeption Context was terminated`
`[bt_action_node-5] [PublisherZMQ] just died. Exeption Socket operation on non-socket`
`[bt_action_node-5] OpenGripper halt`
`[bt_action_node-5] ApproachObject halt`
`[bt_action_node-5] CloseGripper halt`
`[bt_action_node-5] OpenGripper halt`
`[plansys2_node-1] [ERROR] [1624737316.838242090] [executor]: Executor BT finished with FAILURE state`
`[plansys2_node-1] [INFO] [1624737316.838742886] [executor]: Plan Failed`
`[plansys2_node-1] [PublisherZMQ] Server quitting.`
`[plansys2_node-1] [PublisherZMQ] just died. Exeption Context was terminated`
`[plansys2_node-1] [PublisherZMQ] Publisher quitting.`
`[plansys2_node-1] [PublisherZMQ] just died. Exeption Context was terminated`

I have tried small variations on the knowledge and goal but have the same result for any plan combining different actions other than simple moves from one zone to another.

tammerb, 2021-06-26:
The common step where things go wrong in the by_action_node is:
`[bt_action_node-2] [INFO] [1624738478.858006125] [move_1]: Waiting for "navigate_to_pose" action server`
`[bt_action_node-2] [PublisherZMQ] Server quitting.`

fmrico, 2021-06-30:
Hi @tammerb 

This example is designed to interact with Nav2. You can start nav2 in another terminal or use a `nav2_sim_node that implements a "fake" Nav2 (a `navigate_to_pose` action server).

tammerb, 2021-06-30:
@fmrico yes I have the Nav2 node running (the real one) and it works for a simple plan with only a move action. This error only happens when there is transport or assemble actions and then a move action is called after that, as in the output above.

fmrico, 2022-07-09:
Closing... reopen if you continue with this error

closed 2022-07-09 by fmrico