PlanSys2/ros2_planning_system #155: Plan with parallel actions seems to result in failed execution (may be wrong BT is built)

roveri-marco, 2021-08-23:
We have a small example with a PDDL domain where there are some durative actions (e.g. movetoward, doclean, recharge) that cannot be executed in parallel by the same agent (thanks to a workfree predicate checked to be true as precondition at start, set ato false as effect at start, and back to true at end). Then there is another action asksweeping that do not use this mechanism and has the effect to set predicate swept for a certain waypoint. The doclean action for a given waypoint has among others the precondition that the waypoint is already swept. See the `domain.pddl` file in attach. We created the CPP code for the above actions following examples in the distribution. In attach there is the archive psys2_test.zip.

We compile the code with the command:

`colcon build --packages-select psys2_test --symlink-install`

We then run the example as follows:

`ros2 launch psys2_test cleaner.launch.py`

On another terminal, we activate the plansys2 terminal with the command:

`ros2 run  plansys2_terminal plansys2_terminal --ros-args -r __ns:=/cleaner`

At the terminal prompt we then submit the following commands:

```
set instance cleaner robot
set instance sweeper robot
set instance bathroom waypoint
set instance kitchen waypoint
set instance bedroom waypoint
set instance c_dock waypoint

set predicate (recharging_station c_dock)
set predicate (workfree cleaner)
set predicate (issweeper sweeper)
set predicate (in cleaner c_dock)

set function (battery_charge 90)

set goal (and(cleaned bathroom)(cleaned kitchen))

get plan
```

The plan is succesfully computed by popf2 and returned:

```0.000: (movetoward cleaner c_dock bathroom)  [4.000]
0.000: (asksweeping cleaner sweeper bathroom)  [8.000]
0.000: (asksweeping cleaner sweeper kitchen)  [8.000]
8.001: (doclean cleaner bathroom)  [4.000]
12.002: (movetoward cleaner bathroom kitchen)  [4.000]
16.003: (movetoward cleaner kitchen c_dock)  [4.000]
20.004: (recharge cleaner c_dock)  [4.000]
24.005: (movetoward cleaner c_dock kitchen)  [4.000]
28.006: (doclean cleaner kitchen)  [4.000]
```

Here it is clear that the (doclean cleaner bathroom) shall start at 8.001 when the (movetoward cleaner c_dock bathroom), (asksweeping cleaner sweeper bathroom), and (asksweeping cleaner sweeper kitchen) are finished. Indeed, only after the (asksweeping cleaner sweeper bathroom) it is possible to execute the (doclean cleaner bathroom).

If then i run the plan typing in the terminal the command

`run`

The execution of the plan fails. It seems that the (doclean cleaner bathroom) action is started before the (asksweeping cleaner sweeper bathroom) has terminated. Indeed, the asksweeping actions are about at 60% of completion.

Moreover, it seems also that the at end effect (workfree cleaner) of the (movetoward cleaner c_dock bathroom) is first added, and then removed from the problem expert (looking at the logs on the first terminal that print the problem after each update of the problem expert with the action effects).

Looking at the generated bt.xml I see that the bt generated has an initial PARALLEL where the first element is the single action

`(asksweeping cleaner sweeper bathroom)`

the second action in parallel is

`(asksweeping cleaner sweeper kitchen)`

then the third element is the sequence

`(movetoward cleaner c_dock bathroom)`
`(doclean cleaner bathroom)`
`(movetoward cleaner bathroom kitchen)`
`(movetoward cleaner kitchen c_dock)`
`(recharge cleaner c_dock)`
`(movetoward cleaner c_dock kitchen)`
`(doclean cleaner kitchen)`

I see that the (doclean cleaner bathroom) has

```
 <WaitAtStartReq action="(movetoward cleaner c_dock bathroom):0"/>
 <WaitAtStartReq action="(asksweeping cleaner sweeper bathroom):0"/>
```

that it is not clear whether they aim to check for the respective action to terminate before proceeding with this action, or it simply checks for the at start preconditions of the given action to hold.

In the execution it seems that indeed, it checks for the preconditions and thus it starts the next action, in this case the (doclean cleaner bathroom) that indeed seems to fail because its preconditions do not hold.

I believe that checking for the preconditions in this case might be wrong, and will result in failure in the execution.

We did also an experiment by making the asksweeping actions to terminate artificially before the movetoward (changing the
progress_upd for the asksweeping actions to be 0.5 in the launcher), thus having them to terminate before the first movetoward. In this case the execution succeeds.

This suggests that there seem to be something wrong in the way the BT is built from the plan or in the way it is executed.

If on the other end, it is a problem in our modeling, then it would be appreciated to get some feedback.

Finally, it would be nice to get some additional information on the reason a plan failed (i.e. which precondition is not satisfied). This will definitely help the user to debug and understand what's happening.

* PDDL domain: [cleaner-domain.txt](https://github.com/IntelligentRoboticsLabs/ros2_planning_system/files/7031372/cleaner-domain.txt)
* TERMINAL CMD: [cleaner-problem.psys2_terminal.txt](https://github.com/IntelligentRoboticsLabs/ros2_planning_system/files/7031375/cleaner-problem.psys2_terminal.txt)
* bt.xml [bt.txt](https://github.com/IntelligentRoboticsLabs/ros2_planning_system/files/7031379/bt.txt)
* Complete source code of example: [psys2_test.zip](https://github.com/IntelligentRoboticsLabs/ros2_planning_system/files/7031382/psys2_test.zip)

fmrico, 2021-08-30:
@roveri-marco , please, see #156

roveri-marco, 2021-09-10:
I confirm that with these fixes the problem we faced disappear (also in the framework we first identified the problem). For us and if ok with you all, we can merge.

fmrico, 2021-09-10:
Thanks, @roveri-marco

PR #156 (2021-09-10): Fix bt xml
Dear all,

This PR fixes #155 and introduces one change in the resulting BT.

if action `(movetoward cleaner kitchen c_dock):16031` depends on the completion of a previous action `(movetoward cleaner bathroom kitchen):12021`, the resulting BT was:
```
                <Sequence name="(movetoward cleaner kitchen c_dock):16031">
                  <WaitAtStartReq action="(movetoward cleaner bathroom kitchen):12021"/>
                  <ApplyAtStartEffect action="(movetoward cleaner kitchen c_dock):16031"/>
                  <ReactiveSequence name="(movetoward cleaner kitchen c_dock):160 [... cut]
closed 2021-09-10 by fmrico