PlanSys2/ros2_planning_system #157: Get action feedback during running

RicardoM17, 2021-08-30:
What is the recommended way to get feedback on the controller node that a certain has failed/is running/successfully finished etc.

Currently there's only such option for the entire plan but in order to enable more complex tasks this is useful:

https://github.com/IntelligentRoboticsLabs/ros2_planning_system_examples/blob/master/plansys2_bt_example/src/assemble_controller_node.cpp#L145

@fmrico as discussed I am opening this. If you can share any pointers on where to look and on the best way to implement this, that would be great: )

fmrico, 2021-09-01:
Hi Ricardo,

There are some topics you could subscribe to get runtime information:

* `/action_execution_info` [`plansys2_msgs/ActionExecutionInfo.msg`] This topic is dedicated to get the feedback from the action execution (start, and status time, completion, messages, arguments...).
* `/performers_status`[`plansys2_msgs/ActionPerformerStatus.msg`] This topic receives a heartbeat from the action performers, with their metainfo.
* `/actions_hub` [`plansys2_msgs/ActionExecution.msg`] This topic is used to implement the protocol to deliver actions to the action performers. It includes feedback on the completion percentage and the status.

You could subscribe manually, or we could develop a class, to be instantiated in the controllers, to manage all this information. By this class, apart from an API to direct queries, we could set callbacks when certain packages are received. Examples:
* Notify with every completion
* Notify with the completion of certain action
* Notify with all the feedback messages

What do you think?

RicardoM17, 2021-09-13:
Hi @fmrico 

I will have a look at the topics and report back. I think it is a good approach.

Regarding the class I think it is a good idea but I would need a bit of help with that. Not sure if we can work on this together or if there's some examples that you can share so I could have a base to work on.

Let me know what you think is best.

fmrico, 2021-09-16:
Hi @RicardoM17 

Maybe you could do an initial PR with a tentative implementation, and I can review, or even contribute to it.

Best

fmrico, 2026-10-09:
The status of every action in the plan is available from the controller through `ExecutorClient::getFeedBack().action_execution_status`, and also in the plan result. Each `ActionExecutionInfo` carries the action status (`NOT_EXECUTED`, `EXECUTING`, `FAILED`, `SUCCEEDED`, `CANCELLED`), its completion and its status message. `plansys2_patrol_navigation_example/src/patrolling_controller_node.cpp` in the examples repo shows how to use it, and the `action_execution_info` topic is there for asynchronous monitoring. Since this answers the question and there has been no further activity, I'm closing it. Feel free to reopen with a proposal if you would still like a callback-based helper.

Thanks a lot @RicardoM17 for bringing this up, it was a useful question for many users.

closed 2026-10-09 by fmrico