PlanSys2/ros2_planning_system #158: interface package request

yamokosk, 2021-08-30:
I'm really a big fan of the various Client classes. However I feel like there is some dependency leakage with their current implementation. For instance, I am writing a package where I want to use `ExecutorClient`. But `ExecutorClient` is inside `plansys2_executor`. As a result my package now has dependencies on pretty much all of plansys2 as well as other custom repos like `rclcpp_cascade_lifecycle`.

It would be better if the various Client classes were moved to a dedicated `plansys2_interfaces` package or something similar. This would be a big change, so I didn't want to do anything without some feedback from other devs.

fmrico, 2021-09-01:
Hi @yamokosk 

Thank you for your comments. Yes, it would be a big change and I would like to keep it backward compatible. Such a big change would make all users adapt their code. I also can't quite see why it is a problem to depend on other packages from plansys2, or on rclcpp_cascade_lifecycle, which is a package that can be installed using binaries. Perhaps this dependency should be rethought since I do not know if this feature is used by someone.

One option could be to create the package you propose, with the pure virtual base classes (* Interface.hpp) and duplicate the clients there, marking the previous ones as deprecated :thinking: 

More ideas?

fmrico, 2022-07-09:
Hi @yamokosk , I will close this. Reopen if you want to continue discussing...

closed 2022-07-09 by fmrico