PlanSys2/ros2_planning_system #161: Through useful exceptions instead of failing sielently

MostafaGomaa, 2021-09-14:
This is to document places in the code where the program terminates silently instead of throwing a useful exception. This makes it hard to understand what actually happened or to recover in run-time.

MostafaGomaa, 2021-09-14:
https://github.com/IntelligentRoboticsLabs/ros2_planning_system/blob/e713d72a4aa617d5e50439b8afe079d84851e8f2/plansys2_pddl_parser/src/plansys2_pddl_parser/TypeGround.cpp#L28

fmrico, 2021-09-16:
Hi @MostafaGomaa 

This is clearly a bug or a design error. We should address this soon.

Thanks

sarcasticnature, 2022-02-02:
I'm currently working on adding/fleshing out failure messages to some stuff, as part of a PR to fix #174.  So far I have implemented a more descriptive failure message to the `ExecutorClient` when actions fail, but be adding some more runtime failure info to a few other places as well.

closed 2022-07-09 by fmrico