PlanSys2/ros2_planning_system #164: [Q] Are there any plans for further integration with MoveIt2?

solid-sinusoid, 2021-11-16:
Hi guys, at the moment our open source project wants to use your project in the robotic assembly process, but requires the integration of PlanSys2 and MoveIt2, is there such an update planned and can we contribute to this somehow?

For our purposes, there is a need to use PlanSys2 as a replacement for the not yet ported MoveIt Task Constructor. In addition, I can see the advantage of easily parallelizing the actions of the robots, based on your cooking examples.

fmrico, 2021-11-16:
Hi @solid-sinusoid 

Well, the idea is to make PlanSys2 agnostic to another component of any system. In the patrol example, [PlanSys2 uses Nav2 through a generic class](https://github.com/IntelligentRoboticsLabs/ros2_planning_system_examples/blob/2f98d43bf895ace68d1d883ee9638233a4259a7c/plansys2_bt_example/src/behavior_tree_nodes/Move.cpp#L36) able to call any ROS2 action server from a BT node. 

```
Move::Move(
  const std::string & xml_tag_name,
  const std::string & action_name,
  const BT::NodeConfiguration & conf)
: plansys2::BtActionNode<nav2_msgs::action::NavigateToPose>(xml_tag_name, action_name, conf)
{

...

BT_REGISTER_NODES(factory)
{
  BT::NodeBuilder builder =
    [](const std::string & name, const BT::NodeConfiguration & config)
    {
      return std::make_unique<plansys2_bt_tests::Move>(
        name, "navigate_to_pose", config);
    };

  factory.registerBuilder<plansys2_bt_tests::Move>(
    "Move", builder);
}

```

I am not aware of MoveIt2 interface, but MoveIt used ROS actions (among others) to control an arm. Maybe you could use this class...

Best

solid-sinusoid, 2021-11-17:
Thanks for your feedback, I will definitely write if you have any questions.

closed 2021-11-17 by solid-sinusoid