PlanSys2/ros2_planning_system #165: "[terminal]: No problem file specified" while running plansys2 with problem_file option specified

movefasta, 2021-11-17:
Hi

OS: Ubuntu 20.04
Ros2: Foxy
Plansys2: 1.0.10

I tried to run bringup launch with [cooking experiment](https://github.com/IntelligentRoboticsLabs/plansys2_cooking_experiment/tree/main/pddl) domain and problem files:

```
$ ros2 launch plansys2_bringup plansys2_bringup_launch_distributed.py model_file:=/tmp/domain.pddl problem_file:=/tmp/problem.pddl
```

But when i run terminal, i'm see this
```
$ ros2 run plansys2_terminal plansys2_terminal
[INFO] [1637158163.457808354] [terminal]: No problem file specified.
ROS2 Planning System console. Type "quit" to finish
> get problem goal
Goal:
```
Meanwhile i can add problem entities from terminal via problem_expert and it works.
Any solutions for that?

Additionally i got some warnings from start (may be it can cause such behaviour?)
```
[INFO] [launch]: Default logging verbosity is set to INFO
/opt/ros/foxy/share/plansys2_bringup/launch/plansys2_bringup_launch_distributed.py:102: UserWarning: The parameter 'node_name' is deprecated, use 'name' instead
  lifecycle_manager_cmd = Node(
/opt/ros/foxy/share/plansys2_domain_expert/launch/domain_expert_launch.py:42: UserWarning: The parameter 'node_executable' is deprecated, use 'executable' instead
  domain_expert_cmd = Node(
/opt/ros/foxy/share/plansys2_domain_expert/launch/domain_expert_launch.py:42: UserWarning: The parameter 'node_name' is deprecated, use 'name' instead
  domain_expert_cmd = Node(
/opt/ros/foxy/share/plansys2_problem_expert/launch/problem_expert_launch.py:42: UserWarning: The parameter 'node_executable' is deprecated, use 'executable' instead
  domain_expert_cmd = Node(
/opt/ros/foxy/share/plansys2_problem_expert/launch/problem_expert_launch.py:42: UserWarning: The parameter 'node_name' is deprecated, use 'name' instead
  domain_expert_cmd = Node(
/opt/ros/foxy/share/plansys2_planner/launch/planner_launch.py:33: UserWarning: The parameter 'node_executable' is deprecated, use 'executable' instead
  planner_cmd = Node(
/opt/ros/foxy/share/plansys2_planner/launch/planner_launch.py:33: UserWarning: The parameter 'node_name' is deprecated, use 'name' instead
  planner_cmd = Node(
/opt/ros/foxy/share/plansys2_executor/launch/executor_launch.py:46: UserWarning: The parameter 'node_executable' is deprecated, use 'executable' instead
  executor_cmd = Node(
/opt/ros/foxy/share/plansys2_executor/launch/executor_launch.py:46: UserWarning: The parameter 'node_name' is deprecated, use 'name' instead
  executor_cmd = Node(
  ```

Thanks!

fmrico, 2021-11-19:
Hi @movefasta 

Do you need the problem file? What does it contain?

movefasta, 2021-11-19:
> Do you need the problem file? What does it contain?

I am considering a scenario in which a problem file is created by a separate application without a terminal or library functions of problem_expert (as shown in some examples). When I found the problem_file command line option to run plansys2, I decided to check it on files from cooking_experiment. This is not a critical obstacle for me to use plansys2, but rather another possible use case that led me to this bug.

fmrico, 2021-12-15:
Hi @movefasta 

Did you progress with this?

Best

movefasta, 2021-12-15:
> Hi @movefasta
> 
> Did you progress with this?
> 
> Best

Hello!
I tried to load domain.pddl and problem.pddl not through the terminal, but directly into the /tmp directory. Can Plansys2 work this way?

Anyway, after your answer, I didn't try to start the system. I will reopen this issue if it happens again and provide details.

closed 2021-12-15 by movefasta