PlanSys2/ros2_planning_system #166: [bt_action_node-2] error while loading shared libraries: libplansys2_bt_actions.so

Splinter1984, 2021-12-06:
I get the following error when I try to run my own package. What could be the problem with libplansys2_bt_actions.so?
```
[bt_action_node-2] /opt/ros/foxy/lib/plansys2_bt_actions/bt_action_node: error while loading shared libraries: libplansys2_bt_actions.so: cannot open shared object file: No such file or directory
[ERROR] [bt_action_node-2]: process has died [pid 31576, exit code 127, cmd '/opt/ros/foxy/lib/plansys2_bt_actions/bt_action_node --ros-args -r __node:=move_1 -r __ns:=/ --params-file /home/robas_ws/install/rasms_manipulator/share/rasms_manipulator/config/params.yaml --params-file /tmp/launch_params_nlztctyb'].
```

movefasta, 2021-12-06:
It seems like broken deb-package. You should try to install plansys2 `foxy-devel` branch from sources

Splinter1984, 2021-12-07:
Thanks for your reply. I compiled a package from the source of the foxy-devel branch and it worked.

closed 2021-12-07 by Splinter1984

sarcasticnature, 2022-01-22:
I'm still getting this error, on both `foxy-devel` and `master`.  I believe it's due to a missing `${PROJECT_NAME}` in the install section of `plansys2_bt_actions` `CMakeLists.txt`