PlanSys2/ros2_planning_system #167: cannot use plan with numeric-fluents (which is allowed in TFD planner)

edhage, 2021-12-10:
when I use TDF-planner still I get an error "Error parsing PDDL: numeric-fluents does not name a known token" in my domain-expert-node, and also the problem-expert-node.

The reason for this is that 
```auto planner = std::make_shared<plansys2::POPFPlanSolver>();``` is loaded in the domain expert node to check the model, even when I specify the TFD-planner. 

This should be changed. The domain expert should check with the TFD-planner and allow numeric-fluents in that case.

fmrico, 2021-12-15:
Hi @edhage 

The decision to use POPF to check the model was made because it was the only solver that was distributed together with PlanSys2. Of course, what you propose seems totally reasonable.

If you want to work on a PR to solve it, I will tell you how to approach it:
1. Add to `plansys2::PlanSolverBase` a pure virtual `check_domain` method, as in `plansys2 :: POPFPlanSolver`
2. Implement this method in `plansys2::TFDPlanSolver`
3. In `plansys2_params.yaml` there should be a parameter for the domain expert specifying the solver, and depending on it, one or the other will be instantiated in `plansys2::DomainExpertNode::on_configure`

Another option would be to use something generic like the one in the package `plansys2_pddl_parser`, but it would be necessary to evaluate if it is sufficiently general.

In any case, I'll be happy to help you and discuss the details in this issue or in a PR when you get started.

Best!!

roveri-marco, 2022-03-28:
The PR #204 GH-204 should also be able to fix this issue.

sea-bass, 2023-09-19:
Maybe `PlanSolverBase` can expose a ROS service that wraps around this plan validation function.

This would provide a way to not have to instantiate a new planner in the domain expert, but rather wait and call for the service from the planner.

... and with some option (maybe a boolean parameter), we could toggle between the behavior of waiting for a planner to validate the domain vs. creating a POPF instance and validating it locally.

sea-bass, 2023-09-24:
Update, I've put up 2 PRs that should allow for this:

* https://github.com/PlanSys2/ros2_planning_system/pull/275
* https://github.com/PlanSys2/plansys2_tfd_plan_solver/pull/12

Looking forward to getting these fixed up and in a mergeable state!

closed 2026-10-17 by fmrico