PlanSys2/ros2_planning_system #168: not compile DomainExpertNode.cpp

edhage, 2021-12-14:
on line 385 and 343 use request->expression.c_str() iso request->expression

fmrico, 2021-12-15:
Hi @edhage 

Why don't you do a PR with your fix instead of indicating what should be modified? Becoming a contributor to an Open Source project like PlanSys2 is very satisfying :wink:

In case you have not done any PR ever, I indicate the steps:

1. Fork this repository.
2. Where you have this repository cloned in your workspace, make a remote to your fork:

`git remote add upstream <url to your fork>`

3. Make a branch for your PR

`git checkout -b fix_domain_expert`

4. Work on your branch and solve the problem, passing all the tests with `colcon test`
5. Upload this branch to your repo

`git push --set-upstream upstream fix_domain_expert`

6. In your repo you will see the option to make a PR to the repo where you cloned it.
7. Describe your PR well and send it.
8. In that PR you can add more commits simply by commit-push, with the revisions that I make.
9. When accepted, you will be a PlanSys2 contributor forever :wink:

I hope it helps!!

edhage, 2021-12-15:
Dear Francisco,

 

If have not forked any repo before, so thank you for the information. I will investigate this mechanism since I am not such a experienced git-user.

 

I have some questions about PlanSys2, can I reach out to you on this github-email, or do you have another email for me.

 

Greetings, Edward

 

I hope it helps!!

fmrico, 2021-12-21:
Hi @edhage 

Write me to fmrico@gmail.com if you need it.

Best

closed 2022-07-09 by fmrico