PlanSys2/ros2_planning_system #169: Missing Functionality for "Imply" language feature

NmBoyd, 2021-12-29:
If I am understanding everything correctly. It appears that there is currently no support for the `imply` language feature as defined in PDDL 2.1. Which would be quite helpful for some of the task planning problems I am looking to solve. Notably for  action preconditions specified like: `(forall (....) (imply (antecedent_predicate) (consequent_predicate))))`.

When could I perhaps expect support for this feature? Or go about implementing it myself? Help would be appreciated. 

Currently this prevents the creation of desired actions. From what I understand, I would have to create a new condition [here](https://github.com/IntelligentRoboticsLabs/ros2_planning_system/blob/47c4e6f1cccf6e6a5683a75c44ba217f73c8f55a/plansys2_pddl_parser/include/plansys2_pddl_parser/Domain.h) based on a class that contains the parsing logic.

fmrico, 2022-01-07:
Dear @NmBoyd 

I've never used `imply`, and any other contributor never rasked for it before. You'll need to implement it by yourself.

In a quick search, `pddl::parse` in package `plansys2_pddl_parser` does not seem to support `imply so it would be necessary to add support there. 

PDDL plan solver program should support it, that is not clear to me. You can also use a plan solver that supports it. PlanSys2 currently POPF (default), Optics and TFD. If you want to use another, you'll need to create a plugin like the one in `plansys2_popf_plan_solver`.

Best
Francisco

NmBoyd, 2022-01-19:
Sorry for the late reply on this. Thanks for the pointers on this. I do see a direction moving forward with doing this then. 
I've found a temporary work around for now to solve my problem, but I'll try to circle back around to looking at an implementation to this at a later date. Can close this issue for now.

closed 2022-01-19 by NmBoyd

sea-bass, 2023-11-23:
FYI @NmBoyd -- I know it's almost 2 years later but here it goes: https://github.com/PlanSys2/ros2_planning_system/pull/283