PlanSys2/ros2_planning_system #173: Plan Error: std::bad_optional_access

frankonerio, 2022-01-28:
Hello everyone,
please i get this error and my node terminates when a plan is not found. 
Any idea why this is happening?

>>terminate called after throwing an instance of 'std::bad_optional_access'
 >> what():  bad optional access

fmrico, 2022-01-30:
Hi @frankonerio 

What branch and distro are you trying?

frankonerio, 2022-01-30:
hello @fmrico 
I am using the foxy-branch ros2_planning_system package and foxy distro. 
I am using a self created domain and problem file. It generates a plan without any issues,
but in an event of 'plan not found' then i get the error:
"terminate called after throwing an instance of 'std::bad_optional_access'
what(): bad optional access'
. Thank You.

fmrico, 2022-01-30:
In plansys2_terminal, or in your code?

frankonerio, 2022-01-30:
> In plansys2_terminal, or in your code?

in code via a controller node:
#include <plansys2_pddl_parser/Utils.h>

#include <memory>
#include <random>
#include <list>
#include <map>
#include <string>
#include <vector>
#include <iostream>

#include "plansys2_msgs/msg/action_execution_info.hpp"
#include "plansys2_msgs/msg/plan.hpp"

#include "plansys2_domain_expert/DomainExpertClient.hpp"
#include "plansys2_executor/ExecutorClient.hpp"
#include "plansys2_planner/PlannerClient.hpp"
#include "plansys2_problem_expert/ProblemExpertClient.hpp"

#include "rclcpp/rclcpp.hpp"
#include "rclcpp_action/rclcpp_action.hpp"

class Controller : public rclcpp::Node
{
public:
  Controller()
  : rclcpp::Node("behavoir_controller")
  {
  }

  void init()
  {
    domain_expert_ = std::make_shared<plansys2::DomainExpertClient>();
    planner_client_ = std::make_shared<plansys2::PlannerClient>();
    problem_expert_ = std::make_shared<plansys2::ProblemExpertClient>();
    executor_client_ = std::make_shared<plansys2::ExecutorClient>();

    init_knowledge();

    auto domain = domain_expert_->getDomain();
    auto problem = problem_expert_->getProblem();
    auto plan = planner_client_->getPlan(domain, problem);
   

    if (plan.has_value()) {
      std::cout << "Plan found for Goal:" <<
      parser::pddl::toString(problem_expert_->getGoal()) << std::endl;
    }
    
    if (!executor_client_->start_plan_execution(plan.value())) {
      RCLCPP_ERROR(get_logger(), "Error starting a new plan");
    }
  }

  void init_knowledge()
  {
    problem_expert_->addInstance(plansys2::Instance{"getbot", "robot"});
    problem_expert_->addInstance(plansys2::Instance{"entrance", "room"});
    problem_expert_->addInstance(plansys2::Instance{"first_floor", "room"});
    problem_expert_->addInstance(plansys2::Instance{"charging_room", "room"});

    problem_expert_->addPredicate(plansys2::Predicate("(connected entrance first_floor)"));
    problem_expert_->addPredicate(plansys2::Predicate("(connected first_floor entrance)"));

    problem_expert_->addPredicate(plansys2::Predicate("(connected charging_room first_floor)"));
    problem_expert_->addPredicate(plansys2::Predicate("(connected first_floor charging_room)"));

    problem_expert_->addPredicate(plansys2::Predicate("(charging_point_at charging_room)"));
    problem_expert_->addPredicate(plansys2::Predicate("(battery_low getbot)"));
    problem_expert_->addPredicate(plansys2::Predicate("(robot_at getbot entrance)"));

    problem_expert_->setGoal(plansys2::Goal("(and(robot_at getbot first_floor))"));
  }
  void step()
  {
    if (!executor_client_->execute_and_check_plan()) { 
      auto result = executor_client_->getResult();
      if (result.has_value()) {
        if (result.value().success){
          RCLCPP_INFO(get_logger(), "Plan succesfully finished");
        
      } else {
        RCLCPP_ERROR(get_logger(), "Plan failed replanning");
      }
    }else{
      RCLCPP_WARN(get_logger(), "No result for this plan");
    }
    //problem_expert_->clearKnowledge();
    //init_knowledge();
    
    }else {
      std::vector<plansys2::Predicate> predicates = problem_expert_->getPredicates();
    }
  }

private:
  std::shared_ptr<plansys2::DomainExpertClient> domain_expert_;
  std::shared_ptr<plansys2::PlannerClient> planner_client_;
  std::shared_ptr<plansys2::ProblemExpertClient> problem_expert_;
  std::shared_ptr<plansys2::ExecutorClient> executor_client_;

};

int main(int argc, char ** argv)
{
  rclcpp::init(argc, argv);
  auto node = std::make_shared<Controller>();

  node->init();

  rclcpp::Rate rate(5);
  while (rclcpp::ok()) {
    node->step();

    rate.sleep();
    rclcpp::spin_some(node->get_node_base_interface());
  }

  rclcpp::shutdown();

  return 0;
}

frankonerio, 2022-01-30:
> In plansys2_terminal, or in your code?

in code: controller node.
#include <plansys2_pddl_parser/Utils.h>

#include <memory>
#include <random>
#include <list>
#include <map>
#include <string>
#include <vector>
#include <iostream>

#include "plansys2_msgs/msg/action_execution_info.hpp"
#include "plansys2_msgs/msg/plan.hpp"

#include "plansys2_domain_expert/DomainExpertClient.hpp"
#include "plansys2_executor/ExecutorClient.hpp"
#include "plansys2_planner/PlannerClient.hpp"
#include "plansys2_problem_expert/ProblemExpertClient.hpp"

#include "rclcpp/rclcpp.hpp"
#include "rclcpp_action/rclcpp_action.hpp"

class Controller : public rclcpp::Node
{
public:
  Controller()
  : rclcpp::Node("behavoir_controller")
  {
  }

  void init()
  {
    domain_expert_ = std::make_shared<plansys2::DomainExpertClient>();
    planner_client_ = std::make_shared<plansys2::PlannerClient>();
    problem_expert_ = std::make_shared<plansys2::ProblemExpertClient>();
    executor_client_ = std::make_shared<plansys2::ExecutorClient>();

    init_knowledge();

    auto domain = domain_expert_->getDomain();
    auto problem = problem_expert_->getProblem();
    auto plan = planner_client_->getPlan(domain, problem);
   

    if (plan.has_value()) {
      std::cout << "Plan found for Goal:" <<
      parser::pddl::toString(problem_expert_->getGoal()) << std::endl;
    }
    
    if (!executor_client_->start_plan_execution(plan.value())) {
      RCLCPP_ERROR(get_logger(), "Error starting a new plan");
    }
  }

  void init_knowledge()
  {
    problem_expert_->addInstance(plansys2::Instance{"getbot", "robot"});
    problem_expert_->addInstance(plansys2::Instance{"entrance", "room"});
    problem_expert_->addInstance(plansys2::Instance{"first_floor", "room"});
    problem_expert_->addInstance(plansys2::Instance{"charging_room", "room"});

    problem_expert_->addPredicate(plansys2::Predicate("(connected entrance first_floor)"));
    problem_expert_->addPredicate(plansys2::Predicate("(connected first_floor entrance)"));

    problem_expert_->addPredicate(plansys2::Predicate("(connected charging_room first_floor)"));
    problem_expert_->addPredicate(plansys2::Predicate("(connected first_floor charging_room)"));

    problem_expert_->addPredicate(plansys2::Predicate("(charging_point_at charging_room)"));
    problem_expert_->addPredicate(plansys2::Predicate("(battery_low getbot)"));
    problem_expert_->addPredicate(plansys2::Predicate("(robot_at getbot entrance)"));

    problem_expert_->setGoal(plansys2::Goal("(and(robot_at getbot first_floor))"));
  }
  void step()
  {
    if (!executor_client_->execute_and_check_plan()) { 
      auto result = executor_client_->getResult();
      if (result.has_value()) {
        if (result.value().success){
          RCLCPP_INFO(get_logger(), "Plan succesfully finished");
        
      } else {
        RCLCPP_ERROR(get_logger(), "Plan failed replanning");
      }
    }else{
      RCLCPP_WARN(get_logger(), "No result for this plan");
    }
    //problem_expert_->clearKnowledge();
    //init_knowledge();
    
    }else {
      std::vector<plansys2::Predicate> predicates = problem_expert_->getPredicates();
    }
  }

private:
  std::shared_ptr<plansys2::DomainExpertClient> domain_expert_;
  std::shared_ptr<plansys2::PlannerClient> planner_client_;
  std::shared_ptr<plansys2::ProblemExpertClient> problem_expert_;
  std::shared_ptr<plansys2::ExecutorClient> executor_client_;

};

int main(int argc, char ** argv)
{
  rclcpp::init(argc, argv);
  auto node = std::make_shared<Controller>();

  node->init();

  rclcpp::Rate rate(5);
  while (rclcpp::ok()) {
    node->step();

    rate.sleep();
    rclcpp::spin_some(node->get_node_base_interface());
  }

  rclcpp::shutdown();

  return 0;
}

frankonerio, 2022-01-30:
> In plansys2_terminal, or in your code?

In code via controller_node
#include <plansys2_pddl_parser/Utils.h>

#include <memory>
#include <random>
#include <list>
#include <map>
#include <string>
#include <vector>
#include <iostream>

#include "plansys2_msgs/msg/action_execution_info.hpp"
#include "plansys2_msgs/msg/plan.hpp"

#include "plansys2_domain_expert/DomainExpertClient.hpp"
#include "plansys2_executor/ExecutorClient.hpp"
#include "plansys2_planner/PlannerClient.hpp"
#include "plansys2_problem_expert/ProblemExpertClient.hpp"

#include "rclcpp/rclcpp.hpp"
#include "rclcpp_action/rclcpp_action.hpp"

class Controller : public rclcpp::Node
{
public:
  Controller()
  : rclcpp::Node("behavoir_controller")
  {
  }

  void init()
  {
    domain_expert_ = std::make_shared<plansys2::DomainExpertClient>();
    planner_client_ = std::make_shared<plansys2::PlannerClient>();
    problem_expert_ = std::make_shared<plansys2::ProblemExpertClient>();
    executor_client_ = std::make_shared<plansys2::ExecutorClient>();

    init_knowledge();

    auto domain = domain_expert_->getDomain();
    auto problem = problem_expert_->getProblem();
    auto plan = planner_client_->getPlan(domain, problem);
   

    if (plan.has_value()) {
      std::cout << "Plan found for Goal:" <<
      parser::pddl::toString(problem_expert_->getGoal()) << std::endl;
    }
    
    if (!executor_client_->start_plan_execution(plan.value())) {
      RCLCPP_ERROR(get_logger(), "Error starting a new plan");
    }
  }

  void init_knowledge()
  {
    problem_expert_->addInstance(plansys2::Instance{"getbot", "robot"});
    problem_expert_->addInstance(plansys2::Instance{"entrance", "room"});
    problem_expert_->addInstance(plansys2::Instance{"first_floor", "room"});
    problem_expert_->addInstance(plansys2::Instance{"charging_room", "room"});

    problem_expert_->addPredicate(plansys2::Predicate("(connected entrance first_floor)"));
    problem_expert_->addPredicate(plansys2::Predicate("(connected first_floor entrance)"));

    problem_expert_->addPredicate(plansys2::Predicate("(connected charging_room first_floor)"));
    problem_expert_->addPredicate(plansys2::Predicate("(connected first_floor charging_room)"));

    problem_expert_->addPredicate(plansys2::Predicate("(charging_point_at charging_room)"));
    problem_expert_->addPredicate(plansys2::Predicate("(battery_low getbot)"));
    problem_expert_->addPredicate(plansys2::Predicate("(robot_at getbot entrance)"));

    problem_expert_->setGoal(plansys2::Goal("(and(robot_at getbot first_floor))"));
  }
  void step()
  {
    if (!executor_client_->execute_and_check_plan()) { 
      auto result = executor_client_->getResult();
      if (result.has_value()) {
        if (result.value().success){
          RCLCPP_INFO(get_logger(), "Plan succesfully finished");
        
      } else {
        RCLCPP_ERROR(get_logger(), "Plan failed replanning");
      }
    }else{
      RCLCPP_WARN(get_logger(), "No result for this plan");
    }
    //problem_expert_->clearKnowledge();
    //init_knowledge();
    
    }else {
      std::vector<plansys2::Predicate> predicates = problem_expert_->getPredicates();
    }
  }

private:
  std::shared_ptr<plansys2::DomainExpertClient> domain_expert_;
  std::shared_ptr<plansys2::PlannerClient> planner_client_;
  std::shared_ptr<plansys2::ProblemExpertClient> problem_expert_;
  std::shared_ptr<plansys2::ExecutorClient> executor_client_;

};

int main(int argc, char ** argv)
{
  rclcpp::init(argc, argv);
  auto node = std::make_shared<Controller>();

  node->init();

  rclcpp::Rate rate(5);
  while (rclcpp::ok()) {
    node->step();

    rate.sleep();
    rclcpp::spin_some(node->get_node_base_interface());
  }

  rclcpp::shutdown();

  return 0;
}

fmrico, 2022-01-30:
Once quick check I would like you to do. In

```
if (plan.has_value()) {
  std::cout << "Plan found for Goal:" <<
  parser::pddl::toString(problem_expert_->getGoal()) << std::endl;
} 

if (!executor_client_->start_plan_execution(plan.value())) {
  RCLCPP_ERROR(get_logger(), "Error starting a new plan");
}```

The second if should be inside first one, because if `plan.has_value()` is false, you are still trying to use `plan.value()`, as the error indicates.

frankonerio, 2022-01-30:
> Once quick check I would like you to do. In
>
> ```
> if (plan.has_value()) {
>   std::cout << "Plan found for Goal:" <<
>   parser::pddl::toString(problem_expert_->getGoal()) << std::endl;
> }
>
> if [...]

THANK YOU! My Mistake.

closed 2022-01-30 by fmrico