PlanSys2/ros2_planning_system #174: BTActionNode should return BT::NodeStatus::FAILURE instead of throwing std::runtime_error

sarcasticnature, 2022-02-02:
Currently the `BTActionNode` class will throw a runtime error if sending a goal to the ROS2 action server fails or the server rejects the goal.  This should be changed to returning a `BT::NodeStatus::FAILURE` so that the behavior tree can handle the error with recovery behavior(s), instead of simply dying.  If recovery behaviors are not implemented/fail, the `ActionExecutor` will properly register that the action has failed instead of waiting indefinitely for a message on `/actions_hub` that will never come.

Also `BTActionNode` currently does not provided a means for a child class to indicate a failure during execution (only throw a runtime/logic/etc. error in the `on_tick` method), creating the same issue as above.  This should also be changed to allow a `BT::NodeStatus::FAILURE` when necessary.

sarcasticnature, 2022-02-02:
I'm currently working on this issue, but wanted to file this issue to let others know and get any feedback @fmrico might have

fmrico, 2022-02-03:
Hi @sarcasticnature 

You are right!! I think this is a good improvement. Go ahead, please.

Thanks!!

sarcasticnature, 2022-02-03:
I'll also be implementing the (optional) port to specify a server timeout, it looks like it was started but never finished.

closed 2022-03-03 by fmrico