PlanSys2/ros2_planning_system #179: plansys2_tools is not in CI && does not compile in Foxy

sarcasticnature, 2022-02-07:
As stated in the title, tests for `plansys2_tools` are currently not run by CI and should be added to the GitHub workflows.

`rclcpp::Logger::set_level` (on line 57 of LoggerNode.cpp) was added in Galactic and is thus unavailable in Foxy.  Also, using `std::setprecision` without including the `<iomanip>` header seems to work in Galactic but not in Foxy (due to C++XX version?).

@fmrico would you consider removing/changing these two lines of code since it's they're the only thing keeping the package from building in Foxy?  While I understand if `master` is supposed to be the latest and greatest, it does seem like a bit of a shame to break compilation for a whole distro based on such a small change (especially given that the `foxy-devel` branch tends to lag a bit behind).

fmrico, 2022-02-08:
Hi @sarcasticnature 

Since #144, the `main` branch points to Galactic. Indeed, I am thinking of opening a discussion about point to Rolling instead.

No new features will be available in Foxy. `foxy-devel` should only receive bug fixes.

Best

sarcasticnature, 2022-02-08:
Understandable.  The issue regarding CI stands, however

sarcasticnature, 2022-02-09:
looks like it was fixed in one of your recent PRs, closing

closed 2022-02-09 by sarcasticnature

fmrico, 2022-02-09:
Yes!! After you warn me, I included it in the last PRs. Thanks 😉