PlanSys2/ros2_planning_system #184: Possible problem in execution and in BT builder

roveri-marco, 2022-02-09:
We have a small example with a PDDL domain where there are some durative actions (e.g. `pick_up`, `move_carrier`, `deliver`). See the
`domain.pddl` file in attach. We created the CPP code for the above actions following examples in the distribution (we modified the code for `plansys2_simple_example`, and you can find attached also the resulting small code.

We compiled the code as suggested in the tutorial, and the we execute it with the following command.

`ros2 launch assignment assignment_launch.py`

On another terminal, we activate the plansys2 terminal with the command:

`ros2 run plansys2_terminal plansys2_terminal`

Then we specify the following problem

```
set instance robot1 robot
set instance carrier1 carrier
set instance p1 person
set instance p2 person
set instance c1 crate
set instance c2 crate
set instance depot location
set instance l1 location
set instance medicine content
set instance food content

set predicate (at_robot robot1 depot)
set predicate (at_carrier carrier1 depot)
set predicate (at_person p1 l1)
set predicate (at_person p2 l1)
set predicate (at_crate c1 depot)
set predicate (at_crate c2 depot)
set predicate (contain c1 medicine)
set predicate (contain c2 food)
set predicate (available c1)
set predicate (available c2)
set predicate (need p1 medicine)
set predicate (need p2 food)
set predicate (free robot1)
set function (= (capacity carrier1) 4)

set goal (and (have p1 medicine) (have p2 food))
```

If we ask for the generation of a plan with `popf` I get the following plan

```
0       (pick_up robot1 carrier1 c1 depot)      2
2.001   (pick_up robot1 carrier1 c2 depot)      2
4.001   (move_carrier robot1 carrier1 depot l1) 4
8.001   (deliver robot1 carrier1 c2 p2 l1 food) 1
9.002   (deliver robot1 carrier1 c1 p1 l1 medicine) 1
```

If I run it the execution succeeds, but on the first terminal I notice several errors in executing the `(pick_up robot1 carrier1 c2
depot)` and the `(deliver robot1 carrier1 c1 p1 l1 medicine)`. I believe there is a problem in the BT that it has been generated (also reported in attach). Indeed, from what I understand 

`(pick_up robot1 carrier1 c1 depot)` and `(pick_up robot1 carrier1 c2 depot)` are the first actions that start since they do not wait for anything

`(move_carrier robot1 carrier1 depot l1)` waits for `(pick_up robot1 carrier1 c2 depot)`

`(deliver robot1 carrier1 c2 p2 l1 food)` waits for `(pick_up robot1 carrier1 c2 depot)` and `(move_carrier robot1 carrier1 depot l1)`

`(deliver robot1 carrier1 c1 p1 l1 medicine)` waits for `(pick_up robot1 carrier1 c1 depot)` and `(deliver robot1 carrier1 c2 p2 l1 food)`

The action `(pick_up robot1 carrier1 c2 depot)` wrongly do not wait for `(pick_up robot1 carrier1 c1 depot)` to terminate and the execution try to execute it in parallel with `(pick_up robot1 carrier1 c1 depot)` but the preconditions do not hold (the robot is not free) and this result in the error reported in the console although the execution from the terminal seems to proceed correctly! An error is also reported later when executing the second deliver.

In attach there is the output of the first terminal to provide all the informations.

[assignment.zip](https://github.com/IntelligentRoboticsLabs/ros2_planning_system/files/8036279/assignment.zip)
[terminal_commands.txt](https://github.com/IntelligentRoboticsLabs/ros2_planning_system/files/8036283/terminal_commands.txt)
[domain.txt](https://github.com/IntelligentRoboticsLabs/ros2_planning_system/files/8036299/domain.txt)
[OUT.txt](https://github.com/IntelligentRoboticsLabs/ros2_planning_system/files/8036301/OUT.txt)

cc: @fmrico

roveri-marco, 2022-02-10:
Hi @fmrico,

I was confused by the "Error" message. The execution is correct, but I still think that there is something unclear to me (at least) in the constructed BT from the generated plan.

If I interpret correctly the BT, the `(pick_up robot1 carrier1 c2 depot)` is not waiting for the `(pick_up robot1 carrier1 c1 depot)`  to terminate as I was expecting to happen,  and in principle it is possible to start it in parallel with `(pick_up robot1 carrier1 c1 depot)', which in my opinion is wrong since it does not respect the plan structure and the plan causality and the use of the use of a mutual exclusive resource. In my opinion it is wrong that it is activated immediately and it waits for the preconditions to become true, since the action is expected to be executed at time 2 and not immediately. 
If they could have been started together, the planner would have put both of them to start at the same time.

When you say that it correctly waits for the `(free robot1)` to become true since it is a pre-condition, I agree that `(free robot1)` is a precondition, but together with this to properly capture the semantics of the plan generated by the planner as I said it shall wait for the first action to terminate and for the precondition to become true. 

The BT as it is now in principle can decide to let the `(pick_up robot1 carrier1 c2 depot)` (it is in parallel) and then the "DEBUG" message will appear for the other action. 

So my point is that instead of waiting for the precondition to become true, it shall wait for the other action to terminate since there is a resource dependency among these two actions that prevent them to be executed in parallel (as identified by the plan).

We may organize a call to better discuss this point if needed.

fmrico, 2022-02-27:
After videocall with @roveri-marco , I label this as a bug. We should plan a fix or this soon

sarcasticnature, 2022-02-28:
I'd be interested in helping with this issue if needed

fmrico, 2022-03-01:
Sure!!! thanks @sarcasticnature

sarcasticnature, 2022-03-01:
Did your call w/ @roveri-marco yield any suspicions as to where exactly the bug is?  I'll try and start on it this week but may not have a ton of time until the weekend

PR #198 (2022-03-06): Check reqs
Hi

This PR is related to #191 and fixes #184.

When creating the graph, the roots (the actions to initiate an execution flow) are identified. They were not correctly identified, because not all the cases were checked to determine if these roots were compatible. Now, if an action requires a predicate which is an effect of another root, or an action produces an effect which is a requisite of another root, it is not considered a root.

Best
Francisco
closed 2022-03-06 by fmrico