PlanSys2/ros2_planning_system #186: master building problems on galactic

roveri-marco, 2022-02-15:
I've created a fresh checkout of the master, and I've tried to compile it as suggested in the documentation.
The command `rosdep install -y -r -q --from-paths src --ignore-src --rosdistro galactic`
generated the following error
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
plansys2_popf_plan_solver: Cannot locate rosdep definition for [popf]
plansys2_executor: Cannot locate rosdep definition for [rclcpp_cascade_lifecycle]
plansys2_tests: Cannot locate rosdep definition for [rclcpp_cascade_lifecycle]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
```
And then running `colcon build --symlink-install` results in the following error:
```
--- stderr: plansys2_executor
CMake Error at CMakeLists.txt:14 (find_package):
  By not providing "Findrclcpp_cascade_lifecycle.cmake" in CMAKE_MODULE_PATH
  this project has asked CMake to find a package configuration file provided
  by "rclcpp_cascade_lifecycle", but CMake did not find one.

  Could not find a package configuration file provided by
  "rclcpp_cascade_lifecycle" with any of the following names:

    rclcpp_cascade_lifecycleConfig.cmake
    rclcpp_cascade_lifecycle-config.cmake

  Add the installation prefix of "rclcpp_cascade_lifecycle" to
  CMAKE_PREFIX_PATH or set "rclcpp_cascade_lifecycle_DIR" to a directory
  containing one of the above files.  If "rclcpp_cascade_lifecycle" provides
  a separate development package or SDK, be sure it has been installed.

---
Failed   <<< plansys2_executor [1.91s, exited with code 1]
```

I had to manually install the package `sudo apt-get install ros-galactic-rclcpp-cascade-lifecycle` to have the command to succeed!

Moreover while compiling the `plansys2_tools` I've noticed the following warning/errors

```
--- stderr: plansys2_tools
/home/marco/plansys2-ws-galactic/src/ros2_planning_system/plansys2_tools/include/rqt_plansys2_knowledge/KnowledgeTree.hpp:0: Note: No relevant classes found. No output generated.
/home/marco/plansys2-ws-galactic/src/ros2_planning_system/plansys2_tools/include/rqt_plansys2_performers/PerformersTree.hpp:0: Note: No relevant classes found. No output generated.
/home/marco/plansys2-ws-galactic/src/ros2_planning_system/plansys2_tools/include/rqt_plansys2_plan/PlanTree.hpp:0: Note: No relevant classes found. No output generated.
---
```

roveri-marco, 2022-02-15:
I forgot to run the `rosdep update`, thus the `rosdep install -y -r -q --from-paths src --ignore-src --rosdistro galactic` failed, after that the missing dependencies are installed, and then compilation proceeds but the two compilation error...

fmrico, 2022-02-27:
I think we can close this. I will review these errors, but they don't seem to hurt.

closed 2022-02-27 by fmrico