PlanSys2/ros2_planning_system #188: Numeric fluents predicates in :conditions  cause  exception

roveri-marco, 2022-02-19:
If  the PDDL planning domain contains numernic fluents, and the domain contains an action with a numeric predicate  of the form `(=  <expr> <expr>)`  or other operators (e.g. >= <= ...) the PDDL parser throws and exception that then prevents to perform plan generation although the popf or tfd planner are generating a plan for that.

In attach there are two simple domain/problems that show this:

[check_domain.pddl.txt](https://github.com/IntelligentRoboticsLabs/ros2_planning_system/files/8103113/check_domain.pddl.txt)
[check_problem.pddl.txt](https://github.com/IntelligentRoboticsLabs/ros2_planning_system/files/8103114/check_problem.pddl.txt)

Indded, if one tries the following command:
`lib/plansys2_pddl_parser/parser check_domain.pddl.txt check_problem.pddl.txt`
you get the following output:
```
terminate called after throwing an instance of 'parser::pddl::UnknownToken'
  what():   does not name a known token
Aborted
```

Since it is now possible to use numeric fluents, one is also expecting that they are supported in preconditions.

I can help to fix this, but I need a bit of advice on how to extend the parser to handle in the preconditions such kind of numeric predicates (and whenever needed).

roveri-marco, 2022-02-22:
It seems that the only thing that are accepted with the `=` operator are references to parameters e.g. for the action in the example one can write `(= ?rob ?rob)`,.

Moreover, if one declares a constant 
```(:constant
   rrr - robotic_agent
)
```
And then uses it in `(= ?rob rrr)`, the parser complains, as well as if one uses the constant  `rrr` later on in a precondition e.g. `(> (carrying_r rrr) (* (max_capacity) 0))`. 

One possibility would be to handle the `=` as the other operators (e.g. `<`), or depending on the arguments being them numeric or object/parameter to properly construct the respective expression/constant value.

fmrico, 2022-02-27:
Hi @roveri-marco 

If I am not wrong, @jjzapf worked refactoring this part some months ago. I do have not a strong opinion here, and I should analyze in-deep this part (`plansys2_pddl_paser`, if I am not wrong) before being really helpful here. @jjzapf , what is the best way, in your opinion, to fix this?

Best

roveri-marco, 2022-02-28:
Hi @fmrico and @jjzapf 

I can help on this if you give me some advice.

Best
Marco

jjzapf, 2022-02-28:
Hi @roveri-marco,

Thanks for creating this issue. I have also noticed similar behavior. Here is what I have discovered so far.

1. Expressions involving equality are not supported as far as I can tell. This is definitely true of the changes I made. However, I also suspect that it is true of the preexisting parser as well.

My changes mostly concerned representing PDDL constructs as trees using ROS2 messages. The Node.msg file (see link below) lists all the constructs that I added support for in my previous PRs. Note that there does not exist an equality expression type.

https://github.com/IntelligentRoboticsLabs/ros2_planning_system/blob/master/plansys2_msgs/msg/Node.msg#L16

I am not 100% sure we want to add support for equality expressions simply because of the numeric problem associated with checking for equality. Whenever I need to check for equality, I convert the expression to 1 or 2 inequality expressions with an epsilon value. For example, in your domain you could use the following inequalities in place of the eqaulity expression.

(< (carrying_r ?rob) (+ (* (max_capacity) 0) 0.1))
(> (carrying_r ?rob) (- (* (max_capacity) 0) 0.1))

replaces

(= (carrying_r ?rob) (* (max_capacity) 0))

With this modification, the following command should succeed.

ros2 run plansys2_pddl_parser parser check_domain.pddl.txt check_problem.pddl.txt

2. There appears to be a bug in BTBuilder regarding inequality expressions in which the left hand side argument is not a function. That is, we don't properly handle inequality expressions in which the left hand side is an expression rather than a function. See code link below. I think this could be fixed fairly easily. Although it's possible that this limitation may show up in other places. I haven't looked very extensively.

https://github.com/IntelligentRoboticsLabs/ros2_planning_system/blob/master/plansys2_executor/src/plansys2_executor/BTBuilder.cpp#L97

Hope this helps. I might have time to work on the second issue in a week or so. Also happy to assist if you want to take it. :)

roveri-marco, 2022-03-01:
Hi @jjzapf 

Thanks for the clarifications. 

As far as 1 is concerned, I know that one can rewrite an = using >= and <=, i.e. in this case  `(and (<= (carrying_r ?rob) (* (max_capacity) 0)) (>= (carrying_r ?rob) (* (max_capacity) 0))`. However, I believe this is error-prone and very unfriendly for a specifier. Notice that the rewriting with >= and <= w.r.t. your encoding is accurate while your is not fine in general. One remark, this encoding requires two checks instead of 1, and this might impact the search of the planner and the number of operations carried out in the node to monitor the conditions. 

Moreover, if the planner handles equalities in conditions, I strongly believe we need to support it. Indeed, one first validates the model offline with the planner (or at least this is what I do, and I believe also other people do) before integrating with the robot. Please notice that the temporal planners I'm aware of i.e. TFD, optic, and POPF support this feature. There are other temporal planners (e.g. FAPE that have other input formalisms) that also support this kind of specifications.

If we do not want to support this feature in PlanSYS2 (it is a design choice) we need to document the limitations we have in the PDDL support in a more precise way. This will simplify the life of the developer/designer of the PDDL problem by avoiding her/him in considering not supported features. 

Of course, if one extends the support for handling = among expressions this will not only impact the parser but will impact also the BT builder (and maybe more).

I'm not yet in a position to enter efficiently the code, but I can provide support for both issues.

jjzapf, 2022-03-01:
You make some good points @roveri-marco.

Just to clarify, this encoding ...

(and (<= (carrying_r ?rob) (* (max_capacity) 0)) (>= (carrying_r ?rob) (* (max_capacity) 0))

is not the same as the encoding I suggested ...

(and (< (carrying_r ?rob) (+ (* (max_capacity) 0) 0.1)) (> (carrying_r ?rob) (- (* (max_capacity) 0) 0.1))).

Note that I am using an epsilon value of 0.1 to specify an equality range. The reason I like this approach is that the PDDL designer chooses the epsilon value. If we were to support equality expressions in PlanSys2, we would need to decide how to handle the numeric precision of the equality comparison. ... I'm sure this is doable (as you said the planners handle it), but it would need a little bit of thought.

Your comment about supporting all features available in the planner plugins is well taken. That does seem like a fair assumption. Also, documenting the places where we currently do not support all the planner features is definitely a good idea.

I'm not opposed to extending PlanSys2 support for equality expressions. As I said, however, we should be careful about how we handle the precision issue. Any suggestions? Unfortunately, I'm not sure when I will have time to work on this feature. If I find some time, I'll be sure to check back with you.

roveri-marco, 2022-03-01:
Hi @jjzapf ,

I see your point about precision, and I'll buy it, but it is not what you expect when you model the domain, unless you make this design choice exploit and/or well documented. An alternative approach (that I like also) is to consider an internal rewriting of the problem where the `=` is converted into conjunction if disequalities and the epsilon is a parameter one can change at configuration time (similarly as one can modify the planner) via options. 

By the way, it is not only a problem of equality, but also of any relational operator, and I like a uniform approach instead of handling an operator differently from the others.

The place where one should care about the precision is in my opinion the execution and monitoring components where the evaluation of the pre/post conditions takes place. Here one can have an evaluation that interprets the formulas and applies a precision threshold (whose value can be modified via configuration files similarly to the epsilon delay used within the planners). Indeed, the same problem with numeric fluents also happens with the duration of the actions.

With some help, I can try to add the support for the `=`, add the epsilon as a parameter (it would be good to specify it once and use it in different places), and in supporting the epsilon precision in the evaluation. 

Of course, we should first agree on the design choices before committing to performing any modifications.

fmrico, 2022-03-08:
Dear @jjzapf and @roveri-marco 

It seems reasonable to support all PDDL features, including this. I will mark this as "help wanted" waiting for someone to push this.

Thanks for opening this discussion!!

roveri-marco, 2022-03-08:
As soon as I have a bit of time I'll be happy to help with this. Of course, I'll need some support then.

roveri-marco, 2022-03-28:
I've prepared a possible fix in PR #204. Feedback welcome.

roveri-marco, 2022-03-31:
We can close the issue since all the problems solved in GH-204

closed 2022-03-31 by roveri-marco