PlanSys2/ros2_planning_system #199: Alternative Plan-to-BT Algorithm

jjzapf, 2022-03-08:
Hi @fmrico.

I have an idea for an alternative approach to the Plan-to-BT transformation problem.

In the current approach, we first create an action graph by linking action effects to action requirements. We then use the action graph to compute the BT. The action graph approach seems rather complicated. I also wonder if it is solving the right problem. On the surface it appears to be performing a similar function to that of the planner. That is, it appears to be an attempt to determine the correct sequence of actions by linking effects to requirements.

Instead of using the action graph approach, what if we created the BT directly from the plan? To do this, we could define the BT using a single main sequence, where each child would be either (1) an action execution sequence node or (2) a collection of action execution sequence nodes contained within a parallel node.

To illustrate the proposed approach, suppose we have the following partial order plan (taken from “An Introduction to the Planning Domain Definition Language” – Patrick Haslum).

0.000: (move-down e2 n5 n4) [1.000]
0.000: (move-up e1 n1 n2) [1.500]
1.000: (board p3 n4 e2) [2.000]
1.500: (board p1 n2 e1) [2.000]
1.500: (board p2 n2 e1) [3.000]
3.000: (move-down e2 n4 n3) [1.333]
4.334: (move-down e2 n3 n2) [1.333]
4.501: (move-down e1 n2 n1) [1.500]
5.668: (move-down e2 n2 n1) [1.000]
6.001: (leave p2 n1 e1) [3.000]
6.001: (leave p1 n1 e1) [2.000]
6.668: (leave p3 n1 e2) [2.000]

Note that there are 3 examples of concurrency in this plan. That is, there are 3 points in the plan where more than one action can start at the same time. For the example above, the behavior tree would look something like what I've copied below.

Does this make sense? Is there a benefit to the action graph approach not captured by this approach?

I've partially implemented this approach, and it seems to work so far. However, I need to test it on more examples to further verify it. Let me know if you think this is worth pursuing.

Thanks!

    <root main_tree_to_execute="MainTree">
      <BehaviorTree ID="MainTree">
        <Sequence name="MainSequence">
          <Parallel success_threshold="2" failure_threshold="1">
            <Sequence name="(move-down e2 n5 n4):0">
            WAIT_AT_START_ACTIONS
              <ApplyAtStartEffect action="(move-down e2 n5 n4):0"/>
              <ReactiveSequence name="(move-down e2 n5 n4):0">
                <CheckTimeout action="(move-down e2 n5 n4):0"/>
                <CheckOverAllReq action="(move-down e2 n5 n4):0"/>
                <ExecuteAction action="(move-down e2 n5 n4):0"/>
              </ReactiveSequence>
              <CheckAtEndReq action="(move-down e2 n5 n4):0"/>
              <ApplyAtEndEffect action="(move-down e2 n5 n4):0"/>
            </Sequence>
            <Sequence name="(move-up e1 n1 n2):0">
            WAIT_AT_START_ACTIONS
              <ApplyAtStartEffect action="(move-up e1 n1 n2):0"/>
              <ReactiveSequence name="(move-up e1 n1 n2):0">
                <CheckTimeout action="(move-up e1 n1 n2):0"/>
                <CheckOverAllReq action="(move-up e1 n1 n2):0"/>
                <ExecuteAction action="(move-up e1 n1 n2):0"/>
              </ReactiveSequence>
              <CheckAtEndReq action="(move-up e1 n1 n2):0"/>
              <ApplyAtEndEffect action="(move-up e1 n1 n2):0"/>
            </Sequence>
          </Parallel>
          <Sequence name="(board p3 n4 e2):1000">
          WAIT_AT_START_ACTIONS
            <ApplyAtStartEffect action="(board p3 n4 e2):1000"/>
            <ReactiveSequence name="(board p3 n4 e2):1000">
              <CheckTimeout action="(board p3 n4 e2):1000"/>
              <CheckOverAllReq action="(board p3 n4 e2):1000"/>
              <ExecuteAction action="(board p3 n4 e2):1000"/>
            </ReactiveSequence>
            <CheckAtEndReq action="(board p3 n4 e2):1000"/>
            <ApplyAtEndEffect action="(board p3 n4 e2):1000"/>
          </Sequence>
          
          ...
          
        </Sequence>
      </BehaviorTree>
    </root>

roveri-marco, 2022-03-08:
Hi @jjzapf ,

This is interesting.. I need to think about in more detail... but I can only do it from next week!

Marco

fmrico, 2022-03-08:
Hi @jjzapf 

Not always similar times indicate parallel actions. It depends on the duration of each action. Two actions can execute in parallel even if their (int)times are different. What indicates a fork of an execution flow is the completing of an action A followed by some action {B, C} that were waiting for A, but not using the same resources (coded as requirements).

We could avoid creating the graph, but the situation for parallelizing is more complex.

Best

jjzapf, 2022-03-08:
Thanks for the explanation @fmrico. I was just starting to realize this myself by taking a closer look at my example. The first 2 actions involve moving separate elevators (i.e. resources) and the next 3 actions involve loading passengers onto the elevators. The durations of the first 2 move elevator actions are different (1.0 and 1.5). So the loading of passengers is planned to occur at separate times. However, as you said, these events involve separate resources. So, even though they have different start times, they can still overlap (i.e. happen concurrently).

I'll keep mulling this over. Thanks again for the insight!

closed 2022-03-31 by jjzapf