PlanSys2/ros2_planning_system #202: PDDL Parsing: decrease does not name a known token

avekarpp, 2022-03-10:
Hello everyone,
I am fairly new to Plansys2 and was working with the default planner POPF for planning. While planning the domain with PDDL fluents, the PDDL parser and/or (not sure!) domain expert node results in an error for PDDL fluents' decrease effect 

`[ERROR] PDDL Parsing: decrease does not name a known token.` 

The POPF planner is producing a expected plan for the same domain file without any error using command

 `ros2 run popf popf <path/to/domain.pddl> <path/to/problem.pddl>`.

I just introduced a numaric fluent to monitor and modify battery level in the `ros2_planning_system _example/plansys2_simple_example` domain file.

Any help or suggestions are welcome regarding this issue. Thank you in advance!

fmrico, 2022-03-13:
Hi @avekarpp 

Not all the pddl features are supported, even being supported by POPF 😞 

We could mantain this issue open for waiting someone to work on this.

Best

avekarpp, 2022-03-16:
Hello @fmrico,
Thank you very much for your support!

roveri-marco, 2022-03-28:
@avekarpp The problem is that you are mixing PDDL constructs with PDDL+ constructs (indeed, you cannot use the `(decrease (numric_fluent) (* #t  0.5))` which is a feature as far I can understand of PDDL+.

closed 2026-10-19 by fmrico