PlanSys2/ros2_planning_system #205: Goal structure too simple

roveri-marco, 2022-03-28:
Currently, the goal accepted are only of the form '(and (pred1) (pred2)  .... (predN))' however, the planners and a user may want to specify complex expressions (as supported by the planners) including not only simple conjunction of predicates but also a generic formula including fluents and the possibility to have some predicate negated.

roveri-marco, 2022-03-28:
I can work on this and integrate this within the PR GH-204, or create another PR only to solve this issue, but before I would prefer the GH-204 be merged to avoid possible conflicts and to leverage the structure of expressions constructed there.

jjzapf, 2022-03-29:
Looking at the code, I think we are already allowing for a more complex goal than a simple conjunction of predicates.

https://github.com/IntelligentRoboticsLabs/ros2_planning_system/blob/master/plansys2_problem_expert/src/plansys2_problem_expert/ProblemExpert.cpp#L522

Could you describe a little more about what changes you are looking for here?

roveri-marco, 2022-03-29:
Let's consider the file `orig-domain.txt` in attach (replaced for instance in the `plansys2_simple_example`), and the set of commands in the file `comandi.txt` in attach, where there is the goal:
`set goal (and(not(need p1 food))(not(need p2 food))(not(need p3 food))(not(need p4 food))(not(need p1 medecine))(not(need p2 medecine))(not(need p3 medecine))(not(need p4 medecine)))`. In the terminal if I load the commands I correctly see the goal specified:
```
> get problem goal
Goal: (and (not (need p1 food))(not (need p2 food))(not (need p3 food))(not (need p4 food))(not (need p1 medecine))(not (need p2 medecine))(not (need p3 medecine))(not (need p4 medecine)))
> 
```

If I run the command `get plan` I get the answer that no plan exists
```
> get plan
[ERROR] [1648548891.496283076] [planner_client]: /planner/get_plan: Plan not found
Plan not found
> 
```
However, the solution to the problem exists! The reason for this buggy behavior is that the PDDL files generated to submit the problem to the planner are wrong (in particular the instance/problem file is wrong). Indeed, the goal specified there does not contain the negation (see files `gen-domain.txt` and `gen-problem.txt` generated in `/tmp` and attached for convenience)!

`( :goal ( and  ( need p1 food ) ( need p2 food ) ( need p3 food ) ( need p4 food ) ( need p1 medecine ) ( need p2 medecine ) ( need p3 medicine ) ( need p4 medicine ) ) )`  

So something is lost in the flow! It might be because the PDDL parser is not able to handle/parse the correct PDDL file (`orig-problem.txt`). It might be a problem of the problem expert or of the PDDL parser (this for sure has problems) or of both of them.

Attachments:
* [orig-problem.txt](https://github.com/IntelligentRoboticsLabs/ros2_planning_system/files/8370451/orig-problem.txt)
* [orig-domain.txt](https://github.com/IntelligentRoboticsLabs/ros2_planning_system/files/8370454/orig-domain.txt)
* [comandi.txt](https://github.com/IntelligentRoboticsLabs/ros2_planning_system/files/8370464/comandi.txt)
* [gen-dom.txt](https://github.com/IntelligentRoboticsLabs/ros2_planning_system/files/8370459/gen-dom.txt)
* [gen-prob.txt](https://github.com/IntelligentRoboticsLabs/ros2_planning_system/files/8370462/gen-prob.txt)

jjzapf, 2022-03-29:
Thanks for the detailed explanation @roveri-marco. You're right! This does seem to be a current limitation of the system. I think the problem may reside here in the getProblem function.

https://github.com/IntelligentRoboticsLabs/ros2_planning_system/blob/master/plansys2_problem_expert/src/plansys2_problem_expert/ProblemExpert.cpp#L645

It looks like we are currently just extracting the predicates from the goal tree to form the goal part of the problem string.

I think there are 2 potential ways to fix this.

Note that the problem string is created form the instances_, predicates_, functions_, and goal_ objects. These objects are defined from ROS messages as can be seen here.

https://github.com/IntelligentRoboticsLabs/ros2_planning_system/blob/master/plansys2_core/include/plansys2_core/Types.hpp

This was part of the changes I made in #118 to use ROS messages to store and pass around the various PDDL constructs. As part of that PR I added several functions for converting the PDDL constructs to strings.

https://github.com/IntelligentRoboticsLabs/ros2_planning_system/blob/master/plansys2_pddl_parser/include/plansys2_pddl_parser/Utils.h#L83

The instances_, predicates_, functions_, and goal_ objects are currently used to create a parser::pddl::Instance. An << operator overload is then used to extract the string representation of the problem.

**1st  Potential Solution**

Fix the issue concerning conversion from the ROS2 message types to the parser::pddl::Instance type. Hopefully the << operator overload will just work. However, you may need to make some changes to the various PDDLPrint functions.

**2nd Potential Solution**

Don't convert the ROS2 message types to the parser::pddl::Instance type. Instead, just use the string conversion functions in plansys2_pddl_parser/Utils.cpp to construct the problem string.

I'm not sure which approach is the better/easier of the two. Either approach would probably be fine.

roveri-marco, 2022-03-30:
I think that both solutions require to modify the pddl_parser since also there there are limitations (only conjuncts accepted in the goal). I'll think a bit more and I'll try to work on this in the forthcoming days/weeks.

roveri-marco, 2023-01-04:
@fmrico if the issue has been solved and merged, then we may consider closing this issue.

fmrico, 2026-10-09:
This was fixed by #238. The problem expert now writes the full goal tree when it generates the problem, so negated predicates, disjunctions and numeric comparisons in the goal reach the planner unchanged, whether the goal comes from the terminal or from a problem file. Closing as fixed; please reopen if you find a goal construct that still gets lost.

Thank you @roveri-marco for spotting this and for pushing it forward.

closed 2026-10-09 by fmrico