PlanSys2/ros2_planning_system #219: Missing build-dependency for plansys2_tools

roveri-marco, 2022-04-23:
I've tried to compile plansys2 within a Dockerfile and while doing I faced the problem that compilation of plansys2_tools failed with the error
```#7 221.6   "Qt5Widgets", but CMake did not find one.
#7 221.6
#7 221.6   Could not find a package configuration file provided by "Qt5Widgets" with
#7 221.6   any of the following names:
#7 221.6
#7 221.6     Qt5WidgetsConfig.cmake
#7 221.6     qt5widgets-config.cmake
#7 221.6
#7 221.6   Add the installation prefix of "Qt5Widgets" to CMAKE_PREFIX_PATH or set
#7 221.6   "Qt5Widgets_DIR" to a directory containing one of the above files.  If
#7 221.6   "Qt5Widgets" provides a separate development package or SDK, be sure it has
#7 221.6   been installed.
#7 221.6 Call Stack (most recent call first):
#7 221.6   /opt/ros/foxy/share/rqt_gui_cpp/cmake/rqt_gui_cppConfig.cmake:41 (include)
#7 221.6   CMakeLists.txt:10 (find_package)
#7 221.6
#7 221.6
#7 221.6 ---
#7 221.6 Failed   <<< plansys2_tools [1.64s, exited with code 1]
#7 224.9 Aborted  <<< plansys2_planner [4.94s]
```
I had to manually install `qtbase5-dev` `qtbase5-dev-tools` and  `libqt5widgets5`  to successfully compile it.
I believe that adding 
```
  <depend>qtbase5-dev</depend>
  <depend>qtbase5-dev-tools</depend>
  <depend>libqt5widgets5</depend>
```
to the respective `package.xml`  will solve the issue (not yet tried), but for sure this package depends from these ones.

fmrico, 2022-04-26:
Thanks for finding this. May you make a PR with this, please?

roveri-marco, 2022-04-26:
Yes, I'll do later today.

roveri-marco, 2022-04-27:
Fixed with PR #220

closed 2022-04-27 by roveri-marco