PlanSys2/ros2_planning_system #221: Multi-robot example

ana-GT, 2022-04-29:
Hi. I am attempting to test  PlanSys2 for a multi-robot scenario. Is there any tutorial for this particular setup? I have seen the examples for multi-domains and other tutorials, but cannot seem to find an example for multiple robots, specifically, how to run a number of the same action implemented for different robots. From reading a couple of the papers for PlanSys2, it sounds as if I should use the namespace parameter, but i am not quite sure. Thanks!

closed 2022-04-29 by ana-GT

fmrico, 2022-05-03:
Hi @ana-GT 

Currently, the approximation is having a PlanSys2 instance for all robots, and use specialized_arguments 

Good luck ;)

teyssieuman, 2022-05-03:
Hi @ana-GT and @fmrico,

You can find here an example of a multiple-robot scenario:

https://github.com/IntelligentRoboticsLabs/ros2_planning_system/tree/master/plansys2_tests/test_3

Fabrice