PlanSys2/ros2_planning_system #222: plansys2_pddl_parser is broken on Galactic

cottsay, 2022-05-03:
The 2.0.2 release of this repository broke `plansys2_pddl_parser` in Galactic. The package has been broken since April 4th. I believe that b76b33de5e5aae7694bda8decbd5028e05539625 caused the regression.

We're going to sync Galactic to main this week, and unless the package is fixed in the next few days, it will automatically be removed from the Galactic repositories until a fixed version is released, and will need to wait until the next sync to become available again.

fmrico, 2022-05-03:
Hi @cottsay 

Thanks for warning me. I will try to fix it ASAP.

Best

fmrico, 2022-05-03:
[Still failing](https://build.ros2.org/job/Gbin_uF64__plansys2_pddl_parser__ubuntu_focal_amd64__binary/47/console) :sob: 

I thought that removing `find_package(Threads)` and `#include <threads>` could fix it.

The important error messages, IMHO, are:

```
20:35:27 /usr/bin/ld: ./obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp/./obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp/src.c:12: undefined reference to `pthread_detach'
20:35:27 /usr/bin/ld: ./obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp/./obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp/src.c:13: undefined reference to `pthread_join'
. . . 
20:35:27 /usr/bin/ld: cannot find -lpthreads
```

But I am not sure how to fix it, and I am not able to reproduce the error until I create the package and the ROS2 Build farm alerts me of the errors.

Any ideas, @cottsay? Should I try to reproduce in my computer something similar to the RPS2 Build farm to make sure everything is ok before making a PR to rosdistro? 

Francisco

fmrico, 2022-05-03:
I have just found that `<depend>rclcpp</depend>` was missing in `package.xml` in just this package. 

Trying again...

sarcasticnature, 2022-05-03:
it looks like this is the issue now?
```
- Found plansys2_msgs: 2.0.6 (/opt/ros/galactic/share/plansys2_msgs/cmake)
13:21:08 CMake Error at CMakeLists.txt:63 (find_package):
13:21:08   By not providing "Findament_cmake_gtest.cmake" in CMAKE_MODULE_PATH this
13:21:08   project has asked CMake to find a package configuration file provided by
13:21:08   "ament_cmake_gtest", but CMake did not find one.
13:21:08 
13:21:08   Could not find a package configuration file provided by "ament_cmake_gtest"
13:21:08   with any of the following names:
13:21:08 
13:21:08     ament_cmake_gtestConfig.cmake
13:21:08     ament_cmake_gtest-config.cmake
13:21:08 
13:21:08   Add the installation prefix of "ament_cmake_gtest" to CMAKE_PREFIX_PATH or
13:21:08   set "ament_cmake_gtest_DIR" to a directory containing one of the above
13:21:08   files.  If "ament_cmake_gtest" provides a separate development package or
13:21:08   SDK, be sure it has been installed.
```

cottsay, 2022-05-04:
From what I can tell, the `ament_cmake_gtest` error was the first error all along. I'm sorry - my initial analysis about the cause of the regression has misled us.

`ament_cmake_gtest` is referenced here:
https://github.com/IntelligentRoboticsLabs/ros2_planning_system/blob/385d7c61df5b9536f14f9acee6b776b997d7897b/plansys2_pddl_parser/CMakeLists.txt#L63

...but the dependency is missing from the `package.xml`:
https://github.com/IntelligentRoboticsLabs/ros2_planning_system/blob/385d7c61df5b9536f14f9acee6b776b997d7897b/plansys2_pddl_parser/package.xml#L30-L33

Adding the missing dependency should resolve this error.

fmrico, 2022-05-04:
Ok, this fixed the problem.

I have just received an email with the same error in `plansys2_core`. I had the same error. I reviewed all the packages, and I am just preparing a (I hope) last PR to rosdistro to fix everything.

Thanks @cottsay :heart:

cottsay, 2022-05-04:
Looks like the regressions are now resolved. There are still a few packages from this repo building, but I think that the sync is now unblocked. Thanks for the quick action here! We'll now have a regression-free sync of Galactic on Friday.

closed 2022-05-04 by cottsay