PlanSys2/ros2_planning_system #224: Compile error on fresh install

edhage, 2022-05-11:
I installed ROS2 Galactic on Ubuntu 20.04 and did a git clone according to https://intelligentroboticslab.gsyc.urjc.es/ros2_planning_system.github.io/build_instructions/index.html#build

I got this compile error. It seems Plan is not known, and the compiler thinks ‘plansys2_msgs::msg::Plan’ is meant but it does not substitute it, just mentions it:

user@ubuntu:~/plansys_core$ colcon build --symlink-install
Starting >>> plansys2_msgs
Starting >>> plansys2_lifecycle_manager
Finished <<< plansys2_lifecycle_manager [15.2s]                          
[Processing: plansys2_msgs]                              
Finished <<< plansys2_msgs [57.7s]                        
Starting >>> plansys2_pddl_parser
Finished <<< plansys2_pddl_parser [20.7s]                             
Starting >>> plansys2_core
Finished <<< plansys2_core [6.30s]                            
Starting >>> plansys2_popf_plan_solver
Starting >>> plansys2_tfd_plan_solver
--- stderr: plansys2_tfd_plan_solver        
In file included from /home/edward/plansys_core/src/plansys2_tfd_plan_solver/src/plansys2_tfd_plan_solver/tfd_plan_solver.cpp:24:
/home/edward/plansys_core/src/plansys2_tfd_plan_solver/include/plansys2_tfd_plan_solver/tfd_plan_solver.hpp:34:17: error: ‘Plan’ was not declared in this scope; did you mean ‘plansys2_msgs::msg::Plan’?
   34 |   std::optional<Plan> getPlan(
      |                 ^~~~
      |                 plansys2_msgs::msg::Plan
In file included from /home/edward/plansys_core/install/plansys2_msgs/include/plansys2_msgs/msg/plan.hpp:7,
                 from /home/edward/plansys_core/install/plansys2_core/include/plansys2_core/PlanSolverBase.hpp:22,
                 from /home/edward/plansys_core/src/plansys2_tfd_plan_solver/include/plansys2_tfd_plan_solver/tfd_plan_solver.hpp:22,
                 from /home/edward/plansys_core/src/plansys2_tfd_plan_solver/src/plansys2_tfd_plan_solver/tfd_plan_solver.cpp:24:
/home/edward/plansys_core/install/plansys2_msgs/include/plansys2_msgs/msg/detail/plan__struct.hpp:117:7: note: ‘plansys2_msgs::msg::Plan’ declared here
  117 | using Plan =
      |       ^~~~
In file included from /home/edward/plansys_core/src/plansys2_tfd_plan_solver/src/plansys2_tfd_plan_solver/tfd_plan_solver.cpp:24:
/home/edward/plansys_core/src/plansys2_tfd_plan_solver/include/plansys2_tfd_plan_solver/tfd_plan_solver.hpp:34:21: error: template argument 1 is invalid
   34 |   std::optional<Plan> getPlan(
      |                     ^
/home/edward/plansys_core/src/plansys2_tfd_plan_solver/include/plansys2_tfd_plan_solver/tfd_plan_solver.hpp:34:23: error: conflicting return type specified for ‘virtual int plansys2::TFDPlanSolver::getPlan(const string&, const string&, const string&)’
   34 |   std::optional<Plan> getPlan(
      |                       ^~~~~~~
In file included from /home/edward/plansys_core/src/plansys2_tfd_plan_solver/include/plansys2_tfd_plan_solver/tfd_plan_solver.hpp:22,
                 from /home/edward/plansys_core/src/plansys2_tfd_plan_solver/src/plansys2_tfd_plan_solver/tfd_plan_solver.cpp:24:
/home/edward/plansys_core/install/plansys2_core/include/plansys2_core/PlanSolverBase.hpp:39:51: note: overridden function is ‘virtual std::optional<plansys2_msgs::msg::Plan_<std::allocator<void> > > plansys2::PlanSolverBase::getPlan(const string&, const string&, const string&)’
   39 |   virtual std::optional<plansys2_msgs::msg::Plan> getPlan(
      |                                                   ^~~~~~~
/home/edward/plansys_core/src/plansys2_tfd_plan_solver/src/plansys2_tfd_plan_solver/tfd_plan_solver.cpp:52:15: error: ‘Plan’ was not declared in this scope; did you mean ‘plansys2_msgs::msg::Plan’?
   52 | std::optional<Plan>
      |               ^~~~
      |               plansys2_msgs::msg::Plan
In file included from /home/edward/plansys_core/install/plansys2_msgs/include/plansys2_msgs/msg/plan.hpp:7,
                 from /home/edward/plansys_core/install/plansys2_core/include/plansys2_core/PlanSolverBase.hpp:22,
                 from /home/edward/plansys_core/src/plansys2_tfd_plan_solver/include/plansys2_tfd_plan_solver/tfd_plan_solver.hpp:22,
                 from /home/edward/plansys_core/src/plansys2_tfd_plan_solver/src/plansys2_tfd_plan_solver/tfd_plan_solver.cpp:24:
/home/edward/plansys_core/install/plansys2_msgs/include/plansys2_msgs/msg/detail/plan__struct.hpp:117:7: note: ‘plansys2_msgs::msg::Plan’ declared here
  117 | using Plan =
      |       ^~~~
/home/edward/plansys_core/src/plansys2_tfd_plan_solver/src/plansys2_tfd_plan_solver/tfd_plan_solver.cpp:52:19: error: template argument 1 is invalid
   52 | std::optional<Plan>
      |                   ^
/home/edward/plansys_core/src/plansys2_tfd_plan_solver/src/plansys2_tfd_plan_solver/tfd_plan_solver.cpp: In member function ‘virtual int plansys2::TFDPlanSolver::getPlan(const string&, const string&, const string&)’:
/home/edward/plansys_core/src/plansys2_tfd_plan_solver/src/plansys2_tfd_plan_solver/tfd_plan_solver.cpp:64:3: error: ‘Plan’ was not declared in this scope; did you mean ‘plansys2_msgs::msg::Plan’?
   64 |   Plan ret;
      |   ^~~~
      |   plansys2_msgs::msg::Plan
In file included from /home/edward/plansys_core/install/plansys2_msgs/include/plansys2_msgs/msg/plan.hpp:7,
                 from /home/edward/plansys_core/install/plansys2_core/include/plansys2_core/PlanSolverBase.hpp:22,
                 from /home/edward/plansys_core/src/plansys2_tfd_plan_solver/include/plansys2_tfd_plan_solver/tfd_plan_solver.hpp:22,
                 from /home/edward/plansys_core/src/plansys2_tfd_plan_solver/src/plansys2_tfd_plan_solver/tfd_plan_solver.cpp:24:
/home/edward/plansys_core/install/plansys2_msgs/include/plansys2_msgs/msg/detail/plan__struct.hpp:117:7: note: ‘plansys2_msgs::msg::Plan’ declared here
  117 | using Plan =
      |       ^~~~
/home/edward/plansys_core/src/plansys2_tfd_plan_solver/src/plansys2_tfd_plan_solver/tfd_plan_solver.cpp:102:7: error: ‘PlanItem’ was not declared in this scope; did you mean ‘plansys2_msgs::msg::PlanItem’?
  102 |       PlanItem item;
      |       ^~~~~~~~
      |       plansys2_msgs::msg::PlanItem
In file included from /home/edward/plansys_core/install/plansys2_msgs/include/plansys2_msgs/msg/detail/plan__struct.hpp:19,
                 from /home/edward/plansys_core/install/plansys2_msgs/include/plansys2_msgs/msg/plan.hpp:7,
                 from /home/edward/plansys_core/install/plansys2_core/include/plansys2_core/PlanSolverBase.hpp:22,
                 from /home/edward/plansys_core/src/plansys2_tfd_plan_solver/include/plansys2_tfd_plan_solver/tfd_plan_solver.hpp:22,
                 from /home/edward/plansys_core/src/plansys2_tfd_plan_solver/src/plansys2_tfd_plan_solver/tfd_plan_solver.cpp:24:
/home/edward/plansys_core/install/plansys2_msgs/include/plansys2_msgs/msg/detail/plan_item__struct.hpp:149:7: note: ‘plansys2_msgs::msg::PlanItem’ declared here
  149 | using PlanItem =
      |       ^~~~~~~~
/home/edward/plansys_core/src/plansys2_tfd_plan_solver/src/plansys2_tfd_plan_solver/tfd_plan_solver.cpp:112:7: error: ‘item’ was not declared in this scope
  112 |       item.time = std::stof(time);
      |       ^~~~
/home/edward/plansys_core/src/plansys2_tfd_plan_solver/src/plansys2_tfd_plan_solver/tfd_plan_solver.cpp:116:7: error: ‘ret’ was not declared in this scope
  116 |       ret.push_back(item);
      |       ^~~
/home/edward/plansys_core/src/plansys2_tfd_plan_solver/src/plansys2_tfd_plan_solver/tfd_plan_solver.cpp:125:7: error: ‘ret’ was not declared in this scope
  125 |   if (ret.empty()) {
      |       ^~~
make[2]: *** [CMakeFiles/plansys2_tfd_plan_solver.dir/build.make:63: CMakeFiles/plansys2_tfd_plan_solver.dir/src/plansys2_tfd_plan_solver/tfd_plan_solver.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:164: CMakeFiles/plansys2_tfd_plan_solver.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:141: all] Error 2
---
Failed   <<< plansys2_tfd_plan_solver [8.93s, exited with code 2]
Aborted  <<< plansys2_popf_plan_solver [15.4s]

Summary: 4 packages finished [1min 40s]
  1 package failed: plansys2_tfd_plan_solver
  1 package aborted: plansys2_popf_plan_solver
  1 package had stderr output: plansys2_tfd_plan_solver
  14 packages not processed

fmrico, 2022-05-13:
Hi @edhage 

Remove plansys2_tfd_plan_solver from your workspace. I think we need to update the API in that repo.

Thanks!!!

fmrico, 2022-07-09:
I think we can close this

closed 2022-07-09 by fmrico