PlanSys2/ros2_planning_system #227: Consider changing the default behavior of `on_cancelled()`

sarcasticnature, 2022-05-22:
Currently the implementation of `BtActionNode::on_cancelled()` returns `BT::NodeStatus::SUCCESS`.  I have found this to be counter-intuitive, as the goal has (by definition) not been successfully completed, and may often require overriding to return a failure.

I would like to get @fmrico's opinion on changing the default behavior to `BT::NodeStatus::FAILURE` to better reflect what is actually happening.

Splinter1984, 2022-06-09:
If I understand correctly, it would not be better in this case to return BT::NodeStatus::IDLE?

fmrico, 2022-06-10:
Well, the current semantic is `BtActionNode::on_cancelled()` returns  `BT::NodeStatus::SUCCESS` if the action could be succesfully cancelled. I think it is reasonable, isn't it?

fmrico, 2026-10-09:
We're keeping `SUCCESS` as the default of `BtActionNode::on_cancelled()`. It is the same default Nav2 uses in its `BtActionNode` (on Jazzy, Kilted and Rolling), which our class mirrors, and it follows the reasoning given above: a cancellation requested from the tree (for example by a reactive parent) is an expected outcome, not an error. If an action needs to treat a cancellation as a failure, it can override `on_cancelled()` and return `BT::NodeStatus::FAILURE`. Closing, since the behaviour is intended and can be customized per node.

Thanks a lot @sarcasticnature for raising the point, it is a fair question and it is good to have the rationale written down here.

closed 2026-10-09 by fmrico