PlanSys2/ros2_planning_system #258: Problems with types in the problem expert

roveri-marco, 2023-01-17:
The types in the problem expert does not seem to be correct. Indeed, although they are specified correctly in the terminal, the problem.pddl file generated is wrong and causes popf to go in seg fault!

The source code that show the problem: [assignment_3_5.zip](https://github.com/PlanSys2/ros2_planning_system/files/10439515/assignment_3_5.zip)

`ros2 launch assignment_3_5 plansys2_simple_example_launch.py`  on terminal one
`ros2 run plansys2_terminal plansys2_terminal` 
then in the terminal execute the commands
`set instance capacity_4 capacity_number`
`source assignment_3_5/launch/commands`

If you then issue a `get plan` the answer is no plan, but an investigation show that popf goes seg fault since the problem file is not precise.

The generated (correct) domain.pddl file [domain.pddl.txt](https://github.com/PlanSys2/ros2_planning_system/files/10439545/domain.pddl.txt)

The generated (wrong) problem.pddl file 
[problem.pddl.txt](https://github.com/PlanSys2/ros2_planning_system/files/10439563/problem.pddl.txt)

The plan file generated when popf is called 
[plan.txt](https://github.com/PlanSys2/ros2_planning_system/files/10439570/plan.txt)

The commands: 
[commands.txt](https://github.com/PlanSys2/ros2_planning_system/files/10439573/commands.txt)

The correct problem file to generate: 
[good-problem.pddl.txt](https://github.com/PlanSys2/ros2_planning_system/files/10439583/good-problem.pddl.txt)

fmrico, 2023-02-09:
Hi @roveri-marco 

I see that the bug is that it uses the base type instead the derived type of an instance, right?

roveri-marco, 2023-02-10:
Not sure it is only that...

antbono, 2023-05-15:
I tried to use the domain attached, but I can't run on the plansys2 terminal "set predicate ( at dr1 wp4)". Since I have had a similar issue with another simple domain I think the reason is that the problem expert can't understand the hierarchy among types. Could it be?
[domain_logistics.pddl.txt](https://github.com/PlanSys2/ros2_planning_system/files/11479705/domain_logistics.pddl.txt)
[problem_logistics.pddl.txt](https://github.com/PlanSys2/ros2_planning_system/files/11479706/problem_logistics.pddl.txt)

roveri-marco, 2023-05-17:
> 

We can optimize the output to remove the function if no functions declared. It should be possible. I'll take care in one of the possible next pull request.

antbono, 2023-05-17:
> It should work... Can you share the output?

The output of the terminal is:  `Failing to open file "assignment_3_5/launch/commands"`

fmrico, 2026-10-09:
The problem expert writes each instance with the type it was given. In `commands.txt` the deliverables are declared as `deliverable`, so the generated `problem.pddl` matches what was requested, and the constant `capacity_4` is correctly left out of `:objects`. The segfault comes from popf, which crashes when objects are declared with a non-leaf type: the same generated problem is solved fine once the objects use `food`/`tools`. We keep tracking that popf crash in #255. The empty `(:functions)` block in the generated domain mentioned in the comments was fixed in #305. Closing this one in favour of #255.

Thanks @roveri-marco for the files, they made it easy to find the real culprit.

closed 2026-10-09 by fmrico