PlanSys2/ros2_planning_system #259: If predicates have dash in names (e.g. (at-work box1 loc1)) tokenizer of the plansys2_terminal is buggy

roveri-marco, 2023-01-26:
If predicates have dash in names (e.g. (at-work box1 loc1)) tokenizer of the plansys2_terminal is buggy, and it breaks at 
"(at-", "work", "box1", "loc1))" and when reconstructing the string for parsing it build it as (at- work box1 loc1)) adding spaces right after the -, thus creating a wrong token! 
@roveri-marco to add an example and investigate the issue (this is happening while parsing the goal with set goal (and ..)).

Another issue is that the (not or (or in the goal specified in the terminal are wrongly parsed (E.g. it seems the not is lost!) To be further investigated.

fmrico, 2023-02-09:
Good catch!!

fmrico, 2026-10-09:
On current `rolling`, goals with hyphenated names and with `not`/`or` are parsed correctly. For example, `(and (at-work box1 loc1) (not (at-work box2 loc1)))` comes back unchanged, and `(not (or a b))` becomes the equivalent `(and (not a) (not b))`. The unanchored regular expressions in the PDDL parser took the hyphen as a subtraction; they were anchored in #305. Closing as fixed; please reopen with an example if it still happens.

Thank you @roveri-marco for reporting it with such clear examples.

closed 2026-10-09 by fmrico