PlanSys2/ros2_planning_system #260: How to start the system without "clients"

tompe17, 2023-02-01:
I would like to write my own executor that listen on the messages on /actions_hub and takes care of the execution on the robot.  Is that possible?  Is /actions_hub the complete API for this?

The thing I cannot find is how to start the system without some "clients". It seems that the executor is starting something that tries to run the actions.  At least if I start following the first tutorials.  So how do I start it so I can use my own client executor written in Python?

tompe17, 2023-02-02:
I was confused. I now see that the simple example launch file was not the one in bringup so things are working.  So there is no issue with starting the system.

But what I seem to be missing is some unique identifier in the ActionExecution message. How are you supposed to run two action in parallell on the same robot and the same action but maybe different arguments.  Do I have to make some string of the action and arguments so get a unique identifier?

Also what is the field "node_id" in ActionExecution message? What is the meaning?

fmrico, 2023-02-09:
Hi @tompe17 

> I would like to write my own executor that listen on the messages on /actions_hub and takes care of the execution on the robot. Is that possible? Is /actions_hub the complete API for this?

This is the topic that implements the execution auction protocol. You should check all the topics/services/actions that the executor currently has.

I would like to write my own executor that listen on the messages on /actions_hub and takes care of the execution on the robot. Is that possible? Is /actions_hub the complete API for this?

> The thing I cannot find is how to start the system without some "clients". It seems that the executor is starting something that tries to run the actions. At least if I start following the first tutorials. So how do I start it so I can use my own client executor written in Python?

I do not understand this very well. Please, clarify.

> But what I seem to be missing is some unique identifier in the ActionExecution message. How are you supposed to run two action in parallell on the same robot and the same action but maybe different arguments. Do I have to make some string of the action and arguments so get a unique identifier?

For executing the same action in the same robot you should have two different action executors running in the robot, with different node name

> Also what is the field "node_id" in ActionExecution message? What is the meaning?

It is the node name

Francisco

fmrico, 2026-10-09:
The questions here were answered in the thread. `node_id` is the performer's node name, and to run the same action in parallel you start several performers with different node names. If you want to write your own performer in Python, `plansys2_support_py` now provides an `ActionExecutorClient` base class that implements the `actions_hub` protocol. Closing as answered; feel free to open a new issue if something is still unclear.

Thanks @tompe17 for the questions, they helped clarify how the protocol works for everyone.

closed 2026-10-09 by fmrico