PlanSys2/ros2_planning_system #262: Adding inequality check in Actions breaks plansys2-node when executing plan

jeroenzwan, 2023-02-22:
The execution of a plan fails when I add an inequality check to an action in the PDDL domain file, such as
```bash
(at start (> (battery_level ?auv) 10)
```
When I run `get plan` in the plansys2 terminal, the plan is generated correctly. However, when I run `run` in the terminal, the plansys2-node dies with output:
```bash
[ERROR] [plansys2_node-1]: process has died [pid 7302, exit code -11, cmd '/opt/ros/humble/lib/plansys2_bringup/plansys2_node --ros-args -r __ns:=/ --params-file /tmp/launch_params_g_a70tmt --params-file /opt/ros/humble/share/plansys2_bringup/params/plansys2_params.yaml'].
```
I tried running a plan without these inequalities and functions in the condition and effect fields, and that worked without issue. My guess is that it has something to do with the information the action servers obtain from the domain file, since the plan message I send to the executor_client doesn't have weird parameters.

roveri-marco, 2024-03-19:
@jeroenzwan Can you please add examples to reproduce the problem? Of course, we can create one ourselves, but it is a good practice when reporting an issue to also allow to reproduce the bug easily.

fmrico, 2026-10-09:
Numeric comparisons in action conditions, such as `(at start (> (battery_level ?r) 10))`, are supported and covered by the executor tests (for example the charging and road-trip domains), which pass on current branches. We couldn't reproduce the crash without the domain and problem files requested above. Closing for now; please reopen with a minimal domain, problem and launch file if it still happens on a current release.

Thanks @jeroenzwan for taking the time to report it.

closed 2026-10-09 by fmrico