PlanSys2/ros2_planning_system #265: Add terminal instruction to load a problem file

antbono, 2023-05-15:
I was wondering ho to load a pddl problem in plansys2. I found that this feature is already implemented thanks to [#200](https://github.com/PlanSys2/ros2_planning_system/pull/200) but there are no references to it in the tutorial or doc pages. 
I think it could be useful for many user...
I can make a pull request to the documentation but at the moment I don't know what is the right command.
Maybe @roveri-marco or @fmrico can help in this regard.
Thanks!

fmrico, 2023-05-25:
Hi @antbono . Thanks for opening this. I will try to address it soon. I don't know if @roveri-marco can also do it :wink: 

Best

Rezenders, 2024-02-16:
Any updates on this?  I also want to know how to do it but I am not finding an answer. 
Is it possible to load the problem information with a launchfile? like we do for the domain problem?
something like: `ros2 launch plansys2_bringup plansys2_bringup_launch_distributed.py model_file:=/tmp/simple_example.pddl problem_file:=problem_file.pdd` 

I understood that I can load problem info with the plansys2_terminal but it always crash and I have to start a new terminal afterwards:
```
ros2 run plansys2_terminal plansys2_terminal < problem
[INFO] [1708097941.971921837] [terminal]: No problem file specified.
ROS2 Planning System console. Type "quit" to finish
> set instance r2d2 robot
> set instance wp1 waypoint
> set instance wp2 waypoint
> set instance wp3 waypoint
> set instance wp4 waypoint
> set instance wp5 waypoint
> set instance wp6 waypoint
> 
> set predicate (path wp1 wp2)
> set predicate (path wp2 wp3)
> set predicate (path wp3 wp4)
> set predicate (path wp4 wp5)
> set predicate (path wp5 wp6)
> 
> set predicate (robot_at r2d2 wp1)
> [ros2run]: Segmentation fault

```

roveri-marco, 2024-02-16:
Alternatively, on the terminal you can use the `source` command that is expecting a filename as argument, and the file can contains commands to specify the problem as well as all the commands allowed in the terminal.

fmrico, 2024-02-23:
Hi @Rezenders 

Does the `problem_file` launch argument solve your problem? I think I fixed the crash problem in the terminal some time ago :thinking: 

Best

Rezenders, 2024-02-27:
Yes, thanks :)

closed 2024-02-27 by fmrico